riscv/riscv-isa-manual #1154: Clarification regarding trap returns

weiern1997, 2023-10-17:
In section, 3.1.6.1
"_To support nested traps, each privilege mode x that can respond to interrupts has a two-level
stack of interrupt-enable bits and privilege modes. xPIE holds the value of the interrupt-enable bit
active prior to the trap, and xPP holds the previous privilege mode. The xPP fields can only hold
privilege modes up to x, so MPP is two bits wide and SPP is one bit wide. When a trap is taken
from privilege mode y into privilege mode x, xPIE is set to the value of x IE; xIE is set to 0; and
xPP is set to y._"

Suppose my current environment is M-mode, mstatus.MPRV==1 && mstatus.MPP==3.
After taking a trap and executing MRET, the privilege mode will be M, mstatus.MPRV==1 && mstatus.MPP=0.

Given this scenario, previously legal load and store instructions could now be illegal. Is there an issue with my understanding?

aswaterman, 2023-10-17:
Your understanding is correct.  If you take a trap while MPRV=1, the trap handler needs to be quite cautious indeed.

weiern1997, 2023-10-18:
Thank you for your answer! Will be closing this issue now.

closed 2023-10-18 by weiern1997