riscv/riscv-isa-manual #1161: Question about memory order between amo instructions

Tang-Haojin, 2023-11-14:
Hi, I am feeling uncertain about the memory order between two amo instructions. Should two amo instructions in different hart observe sequential memory order, or only need to observe relaxed memory order?

For example, two harts concurrently execute amoswap (with no aq annotation) to swap value x in register with value y in memory. Is it legal to both have value y in register after execution?

aswaterman, 2023-11-15:
The RVWMO spec says that an AMO gives rise to a single memory operation that is both a load and a store, simultaneously.  It isn't correct to think of an AMOSWAP as having a "load phase" and a "store phase".  So, one of the two harts will win the race and perform the entirety of its AMO, making the outcome you mentioned illegal.

Tang-Haojin, 2023-11-15:
Thank you for your kind answer. I understand that load and store of amo operation are atomic.My question is about memory consistency between two harts who both have their private cache. Let me state my question more clearly.

If two harts with private caches concurrently execute amoswap to swap value x in register with value y in memory, is it legal to both return value y in their private caches and store value x in their caches?

Thanks!

aswaterman, 2023-11-15:
No, that would violate RVWMO.

Tang-Haojin, 2023-11-15:
Thank you, I understand now.

Tang-Haojin, 2023-11-15:
Hi, excuse me but we are still uncertain about some of amo instructions' behavior under RVWMO. May I give a specific example to describe my question?

We have two core `P0` and `P1`, where register `x1`s are both `1` initially. Two shared memory, `a` and `b`, are both `0` initially. Then two cores execute following instructions concurrently:

```
 P0                      | P1                      ;
 amoor x0,x1,(a)         | ammor x0,x1,(b)         ;
 ld x2,(b)               | ld x2,(a)               ;
```

Is it legal to both get `x2 == 0`?

The origin of the question is that, if we replace `ammor` with `sd`, we can get `x2 == 0` in both harts if I am not making a mistake, like:

```
 P0                      | P1                      ;
 sd x1,(a)               | sd x1,(b)               ;
 ld x2,(b)               | ld x2,(a)               ;
```

Thus, we are wondering if amo instructions play any special role in RVWMO different from normal memory access instructions.

aswaterman, 2023-11-15:
Nothing prevents these instructions from being reordered, as they access different addresses, have no syntactic dependencies, and have no fences or acquire/release annotations.

Note this is a very different case than your original question, which was fundamentally about atomicity.

Tang-Haojin, 2023-11-16:
> Note this is a very different case than your original question, which was fundamentally about atomicity.

Thank you. That is to say, the case in my original question voilate the _Atomicity Axiom_ in unpriv-isa spec. Do I understand correctly?

closed 2023-11-20 by Tang-Haojin