riscv/riscv-isa-manual #1163: Clarification needed on interrupt delegation

vishmohan, 2023-11-17:
Trying to reconcile the two statements below - 
What is the behavior when _Current Privilege Mode = S , sstatus.SIE = 1 , sip[i] = 1 and sie[i] = 1_?

_Does the machine enter M mode and then delegate the interrupt to S mode if mideleg[i] = 1?_ 
[or]
_Does the machine stay in S mode as indicated in section 4.1.3 ?_

Thank you.

```
From 3.1.8: Machine Trap Delegation registers: (medeleg and mideleg)
By default, all traps at any privilege level are handled in machine mode, though a machine-mode
handler can redirect traps back to the appropriate level with the MRET instruction 

```
```
From 4.1.3 of privileged spec:
An interrupt i will trap to S-mode if both of the following are true: (a) either the current privilege
mode is S and the SIE bit in the sstatus register is set, or the current privilege mode has less
privilege than S-mode; and (b) bit i is set in both sip and sie.
```

aswaterman, 2023-11-18:
There isn't actually a contradiction.  Note the "By default" at the beginning of the first sentence you quoted.   Reading on in 3.1.8, it becomes clear that the "non-default" case is delegation.

The answer to your question is given by that (a) clause in 4.1.3.

poemonsense, 2024-02-04:
I'm having the similar issue here. Thanks in advance if you could help clarify this case @aswaterman 

I assume what this issue concerns is that,

1) According to 3.1.8, this S interrupt (e.g., supervisor timer) is delegated to S-mode ***if the corresponding mideleg bit is set***. This is to say, whether the trap is to M- or S- mode depends on the delegation bit.

2) According to 4.1.3, this S interrupt ***always traps into S-mode regardless of the delegation bit*** when it meets some CSR and priv-mode requirements (a) and (b).

So it seems there is some contradiction here regarding **the dependence on the delegtion bit**.

BTW, I would like to also note:

3) This trap-*to-S-mode* behavior is newly introduced in https://github.com/riscv/riscv-isa-manual/pull/590 and https://github.com/riscv/riscv-isa-manual/pull/701. Before #701, the trap is not explicitly required taken into S-mode (thus by default should depend on the delagtion bit). We are confused about this case because we were jus reading two versions of the spec.

4) Spike looks like following 1) to check the delegation bit as shown in https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/processor.cc#L894. This is also what we assume from the prior version of the RISC-V spec.

poemonsense, 2024-02-04:
> Trying to reconcile the two statements below - What is the behavior when _Current Privilege Mode = S , sstatus.SIE = 1 , sip[i] = 1 and sie[i] = 1_?
>
> _Does the machine enter M mode and then delegate [...]

I think this issue could be stated as follows:

What is the behavior when **mideleg[i] = 0**, Current Privilege Mode = S , sstatus.SIE = 1 , sip[i] = 1 and sie[i] = 1?

poemonsense, 2024-02-06:
> When mideleg[i]=0, sip[i] and sie[i] must be zero (see
> 
> https://github.com/riscv/riscv-isa-manual/blob/704c6d3965336e685b9d3ce351f572cf56fad9c5/src/machine.adoc#L1388
> 
> ), and so the situation you described cannot arise.

Understood. This makes sense. Thanks.