riscv/riscv-isa-manual #1166: Attempting to fetch an instruction from a PMP region that does not have execute permissions raises an instruction access-fault exception.

shruthihk, 2023-11-27:
Hi,
I am not able to see instruction access fault when I disable X bit in pmp configuration to the respective address where I stored the instruction.  Anyone please help on this

shruthihk, 2023-11-28:
Hi ,
 I am trying with spike simulator and in Machine mode . I have loaded the pmpaddr register with some address and in that address I had stored instruction and configure that address by disabling execute (X) bit  then I tried access the instruction which I stored in that instruction.  I am new to this assembly language coding please let me know if my understanding or way of fetching instruction is wrong

allenjbaum, 2023-11-28:
You didn't provide the information that I asked for.
I will venture a guess that you didn't set the corresponding pmpconfig.L bit, therefore the entire address range for that entry will unrestricted by Mmode

shruthihk, 2023-11-29:
Hi
I had set the pmpconfig.L bit to 1 the access succeeds only if the R, W, or X bit corresponding to the access type is set.  When i disable execute bit and try to load address using pseudo instruction "la"

aswaterman, 2025-12-09:
Closed as answered by Allen.