riscv/riscv-isa-manual #1169: Mtimcmp and Stimecmp reset value not declared

ProssMou, 2023-11-30:
In Privileged Spec and Sstc, I cannot find reset value of mtimecmp and stimecmp. What should the reset value be, all f or all 0?

pdonahue-ventana, 2023-11-30:
Like almost everything else, there is no specified reset value.  Because mstatus.MIE is one of the few things that has a specified reset value, we know that we won't take any interrupts immediately out of reset even if mtimecmp and mtime happen to reset to values that cause the interrupt wire to assert.  The M mode code can initialize mtime and mtimecmp to whatever value it wants just like it can initialize the x registers, mideleg, mie, and many other things.  Once mtime and mtimecmp are initialized, mip.MTIP will be in a known state and then the code can clear mstatus.MIE.

allenjbaum, 2023-11-30:
It is a standard definition in the RISC-V architecture world that if a
reset value is not defined in the spec, it is undefined.
That means an implementation can choose any value that it prefers

It was defined this way to give implementations more freedom (I will have
to state personally that I believe this is a big mistake..)
So it could be a random value that might change on every reset (& earn you
the hate of your DV team if you did that....)

It is up to Mmode reset firmware to ensure that CSR values are set such
that there are no spurious events because of architecturally
undefined reset values.
Only the most minimal set of reset values are defined, based on either
security or the possibility they may otherwise interfere with reset Mmode
code.
If there is MMode code that is supposed to be common across vendor
implementations, that will mean a lot more work for it,
though Mmode code usually needs to be personalized per
implementation anyway, so it isn't clear that will be any kind of barrier.

In this particular case, it would be wise to set it to all ones, in my
opinion.