riscv/riscv-isa-manual #1171: Is HFENCE.VVMA, w/wo respect to GVA/ASID/hgatp.VMID, required if G-stage translation for implicit memory access of VS-stage from vsatp is changed?

terranfund, 2023-12-01:
Based on the discussion in 
https://github.com/riscv/riscv-isa-manual/issues/793
I'm still confused that is HFENCE.VVMA, w/wo respect to GVA/ASID/hgatp.VMID, required if G-stage translation for implicit memory access of VS-stage from vsatp is changed?

terranfund, 2023-12-01:
I guess it should be true based on the statement on SPEC RISC-V Privileged Architecture (Document Version 20211203).

> HFENCE.VVMA is valid only in M-mode or HS-mode. Its effect is much the same as temporarily
entering VS-mode and executing SFENCE.VMA. Executing an HFENCE.VVMA guarantees that
any previous stores already visible to the current hart are ordered before all implicit reads by that
hart done for VS-stage address translation for instructions that
 are subsequent to the HFENCE.VVMA, and
 execute when hgatp.VMID has the same setting as it did when HFENCE.VVMA executed.

aswaterman, 2024-10-31:
The ISA spec cannot answer the question you're asking because whether it's "required" depends on what assumptions software is making and what software intends to accomplish by changing a translation.

The spec is clear about what actions HFENCE.GVMA and HFENCE.VVMA perform.

As the note you quoted mentions, if the GVA->GPA translations have not changed, then an HFENCE.VVMA is not needed.  I disagree that there is a contradiction.

If the hypervisor intended to change the GVA->GPA translations as part of some action it takes, then an HFENCE.VVMA would be needed.  But this seems like an exceedingly unlikely scenario.

closed 2024-10-31 by aswaterman