riscv/riscv-isa-manual #1175: Smcntrpmf clarifications

JamesKenneyImperas, 2023-12-14:
Could you please clarify two minor issues about the new Smcntrpmf extension?

1. The description talks about `cycle` and `instret` _counters_. In the Privileged Specification, _CSRs_ with this name are stated to be read-only aliases of the `mcycle` and `minstret` CSRs. Does the extension inhibit update of the M-mode counter views as well? I assume it does.
2. In M-mode, does a CSR write that inhibits M-mode counting get counted or not? Similarly, what about a CSR write that enables M-mode counting?

Thanks.

gfavor, 2023-12-14:
Re #1, since these registers are aliases of each other, then by definition there is only one physical cycle counter and only one physical instret counter (each with two CSR aliases).  Which implies the answer to the question.

Re #2, the spec says:
_For the instret counter, most instructions do not affect mode transitions, so for those the behavior is
clear: instructions that retire in a non-inhibited mode increment instret, and instructions that retire
in an inhibited mode do not._

JamesKenneyImperas, 2023-12-15:
ok thanks

closed 2023-12-15 by JamesKenneyImperas