riscv/riscv-isa-manual #1186: VS-mode interrupts globally disabled in user mode

jrahmeh, 2024-01-05:
There seems to be a discrepancy between sections 9.1 and 9.2.3 of the privileged spec regarding VS-mode interrupts.

Section 9.1 says:
> VS- mode interrupts are globally disabled when executing in U-mode.

And section 9.3:
> An interrupt i will trap to HS-mode whenever all of the following are true: (a) either the current operating mode is HS-mode and the SIE bit in the sstatus register is set, or the current operating mode has less privilege than HS-mode; (b) bit i is set in both sip and sie, or in both hip and hie; and (c) bit i is not set in hideleg.

If privilege mode is U, bit 10 of hip is 1, bit 10 of hie is 1, and bit 10 of hideleg is 0, then an VSE interrupt must be taken according to section 9.2.3.  That contradicts section 9.1

senlinhe, 2025-07-21:
@aswaterman Can you please give a clarification ?

aswaterman, 2025-12-09:
Closing as answered.