riscv/riscv-isa-manual #1191: Legalizing hgatp with an unsupported MODE

YenHaoChen, 2024-01-12:
The spec explicitly states that legalizing hgatp is not the same as doing satp, i.e., ignoring the write. Instead, each hgatp field is WARL individually. However, ignoring write is also a legalization mapping for each CSR field, considering the MODE field.

[image]
[image]

Does a WARL field generally mean the legalization should not consider other fields? Or, can a write to hgatp with an unsupported MODE value be ignored as it is for satp?

gfavor, 2024-01-12:
On Thu, Jan 11, 2024 at 7:37 PM YenHaoChen ***@***.***> wrote:

> The spec explicitly states that legalizing hgatp is not the same as doing
> satp, i.e., ignoring the write. Instead, each hgatp field is WARL
> individually. However, ignoring write is also a [...]
The definition of WARL in the Priv spec says "Implementations can return
any legal value on the read of a WARL field when the last write was of an
illegal value, but the legal value returned should deterministically depend
on the illegal written value and the architectural state of the hart."

> Or, can a write to hgatp with an unsupported MODE value be ignored as it
> is for satp?
>
If ignoring a write results in a later read not returning a legal value per
the above definition, then that would be a non-compliant implementation.

And note that the satp MODE definition, by explicitly saying what it says,
is establishing a "custom" WARL definition for just the satp.MODE field.
This says nothing about any other satp fields or about any WARL fields in
any other CSRs.  By default, the default definition of WARL applies to
'WARL' fields.  But a given 'WARL' CSR field is allowed to be spec'ed
differently (as was done for satp.MODE).

Greg

Message ID: ***@***.***>
>

allenjbaum, 2024-01-13:
There a bit of trickiness there.
Some CSR fields have dependencies on other fields, e.g. PMP RWX fields
don't allow W=1, R=0, so if both are set, and you do a CSRRC to the R-bit,
That's illegal, even though setting that bit in isolation is supposed to be
legal. (that's a case where the RWX bits must be viewed as a single field
even though they are described individually))
There are also cases where setting some bit in one CSR is perfectly legal,
but it causes some field in a different CSR field to become illegal.

aswaterman, 2024-01-13:
Although I agree with your general observation, @allenjbaum, I can't see how it relates to the original post, so I'm closing the issue.

YenHaoChen, 2024-01-15:
Thanks for all of the comments. The final conclusion is:
1) a WARL field could consider other fields, and
2) a write to hgatp with an unsupported MODE value can be ignored as it is for satp.

aswaterman, 2024-01-15:
@YenHaoChen I don't believe your second conclusion is correct.  The spec draws a special case for `satp`, and it does not draw the same special case for `hgatp`.  (This is basically a restatement of what @gfavor wrote earlier.)

YenHaoChen, 2024-01-15:
> If ignoring a write results in a later read not returning a legal value per the above definition, then that would be a non-compliant implementation.

Ignoring a write results in a later read always returning a legal value, given that the previous value is legal.

> And note that the satp MODE definition, by explicitly saying what it says,
is establishing a "custom" WARL definition for just the satp.MODE field.
This says nothing about any other satp fields or about any WARL fields in
any other CSRs.  By default, the default definition of WARL applies to
'WARL' fields.  But a given 'WARL' CSR field is allowed to be spec'ed
differently (as was done for satp.MODE).

I agreed that the satp.MODE has a special WARL.

> The definition of WARL in the Priv spec says "Implementations can return
any legal value on the read of a WARL field when the last write was of an
illegal value, but the legal value returned should deterministically depend
on the illegal written value and the architectural state of the hart."

I may have misled my question. Please allow me to rephrase my question. Does a WARL field generally mean the legalization should not consider **values of other written fields**? If not, the special WARL definition of satp.MODE seems to conform to the general definition of WARL in the spec. Then, why cannot another WARL field, i.e., hgatp.MODE, uses the WARL definition of satp.MODE?

gfavor, 2024-01-15:
On Mon, Jan 15, 2024 at 1:27 AM YenHaoChen ***@***.***> wrote:

> If ignoring a write results in a later read not returning a legal value
> per the above definition, then that would be a non-compliant implementation.
>
> Ignoring a write results in a later read always returning a legal value,
> given that the previous value is legal.
>
The special statement made about satp.MODE - "if satp is written with an
unsupported MODE, the entire write has no effect" - does NOT conform to the
standard WARL definition.

So, for example, if one conforms to the standard definition, then if one
writes an unsupported/illegal MODE value and a legal ASID value, then:

   - Yes, one can ignore the write to MODE.  A later read of MODE will
   still be required to return a legal value (either the existing value if
   that is legal, or some legal value if the current state is 'X' because it
   was not initialized by hardware reset and was left to be initialized by
   software).

   - No, one cannot ignore the write to ASID.  A later read of ASID must
   return this written legal value.

So, since hgatp.MODE is a standard WARL field (without the
specified special satp.MODE behavior), one cannot ignore the entire write
to hgatp if MODE is being written with an illegal value.

Greg

Message ID: ***@***.***>
>

YenHaoChen, 2024-01-16:
Thank you.

The difference between hgatp.ASID (normal WARL) and satp.ASID is that:
- One can determine whether a hgatp.ASID value is supported by writing the value and then reading to see if the value was retained, regardless of written values of other hgatp fields of the same CSR write instruction. 
- On the other hand, one must determine whether a satp.ASID value is supported by writing the value and a valid satp.MODE value within a CSR write instruction.

[image]