riscv/riscv-isa-manual #1193: CSRs size/width for RV64/128 and debug spec register abstract commands

algrobman, 2024-01-15:
Hi

I'm wondering what is actual meaning for CSR width/size in RISCV architecture? 
What should  upper bits of target GPR be after read from "32 bit" CSRs  in RV64? ( I guess 0)
Why are some CSRs defined as 32 bit?  Why can't  these CSRs be defined  XLEN wide  with reserved upper bits? 
Looks like for processor RTL implementation point of view such redefinition changes nothing (csrr* instructions are XLEN wide operations  anyway). 
It seems to me CSRs width difference only matter for register abstract command in debug spec ( which requires to fail such command if access size is higher than "actual" register size), making debugger implementation unnecessary complex, especially to support  privet CSRs, defined as 32 bit . 

Can we change this in future spec version, defining that CSRs are always current XLEN bits wide ?

gfavor, 2024-01-15:
Yes, a 32-bit CSR is zero-extended in RV64.

A full explanation to your general question would be rather long.  But I'll just note that there are important subtleties between spec'ing a CSR as 32b, 64b, XLEN, UXLEN, SXLEN or MXLEN wide.  For example, some CSRs need to be *XLEN wide while others more properly need to be XLEN wide.  All this matters when one has mixed-XLEN systems

algrobman, 2024-01-15:
Since all CSRs need to be accessed in M-mode and M-mode code can change *XLEN to maximum (DXLEN) why can't all CSRs be defined as current XLEN wide? What is the real need to define CSRs  like  DCSR or FCSR as 32bit  for RV64 CPU? (BTW, width of FFLAGS or FRM are not defined) . Or @gfavor ,  are you saying  that some CSRs can be wider that current XLEN?

gfavor, 2024-01-15:
On Mon, Jan 15, 2024 at 11:11 AM algrobman ***@***.***> wrote:

> Since all CSRs need to be accessed in M-mode and M-mode code can change
> *XLEN to maximum (DXLEN) why can't all CSRs be defined as current XLEN
> wide? What is the real need to define CSRs like DCSR or [...]
Some CSRs have a width that is always *XLEN independent of the current XLEN
(and this may even dynamically vary if the associated *XL field is
changed), while others have a width that varies and is always the current
XLEN.  (Btw, note that the ISA specs have a section after "CSR Width
Modulation" for when the width of a CSR changes.)

I apologize but I can't afford to write up a full tutorial about CSR
widths, the various widths possible, what happens in mixed XLEN systems,
and what happens when a given CSR changes in width.  And the why behind all
that.  This is all about corner case and special case behaviors that may
not be obvious to the casual reader of the specs (and of course about what
happens in mixed-XLEN systems).  More detailed and careful reading, along
with consideration of what wants to happen in mixed-XLEN scenarios, is
necessary to figure out what is the proper width specification (and why).

Greg

Message ID: ***@***.***>
>

algrobman, 2024-01-15:
Again, why can't we redefine the 32bit CSRs as XLEN wide? 

We have a few privet CSRs with 1- 2 or no bits  defined. - Now we have to tell debuggers that these are 32 bit (?) This makes debugger developments life more complicated than needed and adds gates to detect wrong access size for abstract commands ...

Don't we fall into the same trap with 64 bit CSRs in RV128 system? Why do the PMP CSRs and satp  support only 64 bit address? - Does this mean that RV128 CPUs should have only 64 bit address path?

algrobman, 2024-01-16:
> Generally CSRs would be defined to be 32 bits in width if they are expected
> to never hold more than 32 bits of fields now and into the future, i.e.
> there's no need for the CSR to ever be 64 bits wide (other than simply
> being zero-extended from 32 bits in RV64).

Why does this matter architecturally? - the CSRs are accessed by 'csrrw* ' instructions which do XLEN size access anyway.
But this "32 bit" semantics causes unnecessary complications in debug module and debuggers. 

In Debug mode all CSRs of RV32 CPUs  are accessed with arrsize = 2 (as 32 bits), but  most RV64 CPU CSRs should be accessed with arrsize=3 and only very few with arrsize=2 ! :( 

I guess RV128 systems will be even more complex to handle in debug mode. 

BTW, different CSR fields layouts  for CSRs is also the problem to handle in debugger software. It needs to keep 3 different maps for such registers to display their fields values. For what?

aswaterman, 2025-12-09:
Closing, as Greg answered the original post.