riscv/riscv-isa-manual #1195: Zfa fround handling of NaN

davidharrishmc, 2024-01-22:
The Zfa spec for fround.* and froundnx.* describes how to handle Zero and Infinite inputs, but is silent about NaN.

The Zfa spec indicates these instructions implement the IEEE754-2019 roundToIntegral/Exact family of instructions, so if the input is a NaN, the output should be a NaN.

The Zfa spec is silent about what kind of NaN is produced.  Sail produces a canonical NaN.

I suggest clarifying the riscv_zfa spec by adding a sentence to 26.3 after "Zero and infinite inputs are copied to rd unmodified.":

**"If the input is a NaN, the output is a canonical NaN."**
----

[31] [M]: 0x000000008000300E (0xF0028053) fmv.w.x ft0, t0
f0 <- 0xFFFFFFFF7FC00001
mem[X,0x0000000080003012] -> 0x70D3
mem[X,0x0000000080003014] -> 0x4040
[32] [M]: 0x0000000080003012 (0x404070D3) fround.s ft1, ft0, dyn
f1 <- 0xFFFFFFFF7FC00000
mem[X,0x0000000080003016] -> 0xD06F
mem[X,0x0000000080003018] -> 0x83CF

gfavor, 2024-01-22:
I think one has to go back to the F extension (that D and Zfa are built
upon), where it says "Except when otherwise stated, if the result of a
floating-point operation is NaN, it is the canonical NaN. The canonical NaN
has a positive sign and all significand bits clear except the MSB, a.k.a.
the quiet bit."  (And the SAIL code matches this.)

Greg

davidharrishmc, 2024-01-23:
I did look at that, but I also saw that FMIN, FMAX, FCVT, FSGNJ*, FMV, FEQ, FLT, FLE and FCLASS specify behavior if an input is NAN.  The Zfa spec also describes Zero, infinite, and signaling NaN. There's just enough uncertainty that I felt a need to run Sail to verify the canonical NaN output.  I agree that the spec isn't wrong, but I think being explicit would save other people wondering.

aswaterman, 2024-01-24:
For my part, I'd prefer to continue to fall back to the "except when otherwise stated" clause, on the basis that I don't want to spell it out for other instructions where the default behavior applies.

davidharrishmc, 2024-01-26:
Thank you for the feedback.  Sounds like time to close this.

closed 2024-01-26 by davidharrishmc