riscv/riscv-isa-manual #1197: Should the exclusive monitor for supporting the lr/sc instruction pair be placed within the RISC-V CPU subsystem?

jiamenghan, 2024-01-22:
Hello, while trying to understand the RISC-V lr/sc instruction pair, I notice that they can be used in non-cacheable or IO regions. However, when employing lr/sc instruction pair in these regions, an exclusive monitor must be added for each thread to monitor different threads' access operations on the same address, thus determining whether the sc instruction can be successfully executed. The RISC-V Specification, however, does not specify where such an exclusive monitor should reside nor does it explicitly mention the existence of the exclusive monitor.

So here's my question: In a RISC-V based SoC system, is there a clear requirement that the exclusive monitor must be placed within the RISC-V CPU subsystem (as indicated by Number 1 in the diagram), or should it reside outside the RISC-V CPU subsystem, perhaps at the Point of Serialization (Number 2 in the diagram), or at the endpoint (Number 3 in the diagram), or is this implementation-defined?
[image]

gfavor, 2024-01-22:
On Mon, Jan 22, 2024 at 4:47 AM jiamenghan ***@***.***> wrote:

> Hello, while trying to understand the RISC-V lr/sc instruction pair, I
> notice that they can be used in non-cacheable or IO regions. However, when
> employing lr/sc instruction pair in these regions, an exclusive monitor
> must be added for each thread to monitor different threads' access
> operations on the same address, thus determining whether the sc instruction
> can be successfully executed. The RISC-V Specification, however, does not
> specify where such an exclusive monitor should reside nor does it
> explicitly mention the existence of the exclusive monitor.
>
> So here's my question: In a RISC-V based SoC system, is there a clear
> requirement that the exclusive monitor must be placed within the RISC-V CPU
> subsystem (as indicated by Number 1 in the diagram), or should it reside
> outside the RISC-V CPU subsystem, perhaps at the Point of Serialization
> (Number 2 in the diagram), or at the endpoint (Number 3 in the diagram), or
> is this implementation-defined?
> 1.jpg (view on web)
> <https://github.com/riscv/riscv-isa-manual/assets/13928390/7cfe59d1-031a-4f21-8c78-04796a157eef>
>
A few initial comments:

   - RISC-V does not require that LR/SC be supported for non-main memory
   PMA regions.

   - The Priv spec PMA chapter describes the Reservability PMA.  This
   section also contains the following non-normative comment:

*We recommend providing RsrvEventual support for main memory regions where
   possible. Most I/O regions will not support LR/SC accesses, as these are
   most conveniently built on top of a cache-coherence scheme, but some may
   support RsrvNonEventual or RsrvEventual.    *

As far as how one would implement LR/SC support for non-cacheable and I/O
regions, the architecture of course avoids specifying implementation
approaches.  So there is no requirement as to where an exclusive monitor
must be placed., i.e. this is implementation-defined.

Greg

jiamenghan, 2024-01-23:
> On Mon, Jan 22, 2024 at 4:47 AM jiamenghan ***@***.***> wrote: Hello, while trying to understand the RISC-V lr/sc instruction pair, I notice that they can be used in non-cacheable or IO regions. [...]

Thx a lot!

closed 2024-01-23 by jiamenghan