riscv/riscv-isa-manual #1198: A standard extension: chapiter 8.2, missing "valid reservation set" definition

sylware, 2024-01-22:
It says "the SC.W succeeds only if the reservation is still valid".

"valid" needs a formal definition/clarification has it is required/critical for LR/SC implementation.
(values of bytes? any kind of access from anywhere? etc)

gfavor, 2024-01-22:
Whether a reservation is valid or not corresponds to the spec's description of when an SC may succeed and when an SC must fail, i.e. the success conditions describe what is necessary to have a valid reservation, and the failure conditions correspond to conditions that invalidate a reservation.

sylware, 2024-01-23:
How "preemption" is supposed to know a hart has currently a reservation set and to know where(address, old virtual? new virtual? physical?) to "forcibly" generate an observed write to such reservation set in order to invalidate it?

(trap instruction? some control registers? I missed something?)

closed 2024-01-23 by sylware

sylware, 2024-01-23:
oops, close the issue.

I read a post on stack exchange presenting how linux does it.

Basically, we have to be _sure_ to perform a sc instruction from a _different_ address (linux does a load probably to help architecture supporting "constraint lr/sc loops").