riscv/riscv-isa-manual #1201: alloy.mit.edu link is dead

kbroch-rivosinc, 2024-01-25:
https://github.com/riscv/riscv-isa-manual/blob/97f0cf734ca244fe4cca8edbe366a34ad09e5f89/src/mm-alloy.adoc?plain=1#L5

```
kbroch@~/000Chromeos-Downloads via 🐍 v3.11.2 (venv) 
❯ curl --output /dev/null  --write-out "%{http_code}" "https://alloy.mit.edu"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (7) Failed to connect to alloy.mit.edu port 443 after 391 ms: Couldn't connect to server
000
kbroch@~/000Chromeos-Downloads via 🐍 v3.11.2 (venv) 
❯ curl --output /dev/null  --write-out "%{http_code}" "http://alloy.mit.edu"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (7) Failed to connect to alloy.mit.edu port 80 after 1801 ms: Couldn't connect to server
000
```

possible new homes?: 

* https://www.csail.mit.edu/research/alloy
* https://alloytools.org/

wmat, 2024-01-25:
Both of those replacements seem valid but for sake of brevity, https://alloytools.org/ seems like a logical replacement.  Feel free to shoot a PR for the fix.

wmat, 2024-01-25:
Thanks @kbroch-rivosinc

closed 2024-01-25 by wmat