riscv/riscv-isa-manual #1205: Multi-copy atomic vs other-multi-copy atomic

IEncinas10, 2024-01-28:
Hello. While reading the spec, I found a particular sentence to be confusing: https://github.com/riscv/riscv-isa-manual/blob/72ff2f6468c7f7de209ce7e037a5303a18235ce6/src/mm-eplan.adoc?plain=1#L672 

Which is later clarified in the same file:

```
Both RVWMO and RVTSO are multi-copy atomic (or _other-multi-copy-atomic_): 
any store value that is visible to a hart other than the one that originally 
issued it must also be conceptually visible to all other harts in the system.
```

From my understanding, `multi-copy` is not the same as `other-multi-copy-atomicity`. Although I find the complete discussion regarding this topic in the manual to be clear, the isolated first sentence stating `Finally, we note that since RISC-V uses a multi-copy atomic memory` might be confusing (particularly if only that sentence is read, and not the whole text).

Wouldn't it be more precise to describe the model as `other-multi-copy-atomic` instead of `multi-copy atomic`?

gfavor, 2024-01-28:
Common terminology (before ARMv8 came along with its new-ish term) was along the following lines:

• Single-copy atomic: a store becomes visible to all processors at the same time (e.g. SC).
• Multi-copy atomic: a store becomes visible to the issuing processor before becoming visible simultaneously to all other processors (e.g. in , TSO, and Alpha).
• Non-atomic (or non-multi-copy-atomic): a store becomes visible to different processors at different times (e.g. POWER).

ARM, for better or worse (depending on your perspective) introduced the new and more self-clarifying term “other multi-copy-atomic” - which corresponds to the common term “multicopy-atomic” as used now a days.  In other words, RISC-V uses the common term that most architectures use (but provides the parenthetical you noted for people coming from ARMv8 architecture).  Only ARMv8 formally uses the different and "better" version of the term.

IEncinas10, 2024-01-28:
Thanks for the clarification! 

Then, if I'm understanding this correctly:

- ARM's multi-copy atomic  == Single-copy atomic
- ARM's other-multi-copy atomic == Multi-copy atomic

I guess that if they have redefined the term `multi-copy atomic` with a different term it is on them, but for clarity's sake, could the first `multi-copy atomic` point to the definition of the term? 

If this has already been discussed in the past feel free to close the issue.

gfavor, 2024-01-29:
Your second bullet is correct (ARM's term is just trying to be more self-descriptive than the broadly used term).  Everyone essentially means the same thing.

Your first bullet is not applicable to ARMv8 since it only uses the term other-multi-copy atomic (and is not correct in general).

As far as pointing to a definition of "multi-copy atomic", this is an established industry term and RISC-V specs in general don't try to point to definitions of all the industry standard terms that are used in the specs.

gfavor, 2024-01-29:
Nothing more to be said.

closed 2024-01-29 by gfavor