riscv/riscv-isa-manual #1221: Reminder: Add short acronym names of CSRs into smstateen integrated chapter.

wmat, 2024-02-07:
This issue serves as a reminder to add the short acronyms of CSRs into the Smstateen integrated chapter as found in [PR](https://github.com/riscv/riscv-isa-manual/pull/1218)

wmat, 2024-02-29:
@gfavor would you be able to add these shart names to the PR for smstateen?

gfavor, 2024-02-29:
@bill traynor ***@***.***> , could you do this?  (I'm not going to be
in a position for a while to do this.). Or do you need help/guidance as to
exactly how/where to incorporate the acronym names?

wmat, 2024-02-29:
@gfavor OK, I think I can take a shot as there only seems to be one place they can go.

wmat, 2024-02-29:
Actually, I'm not comfortable adding this content.  @ved-rivos is this something you could add to smstateen.adoc?  Here's the section it goes in, I just am not sure where it goes:

Initially, the following bits are defined in `mstateen0`,
`hstateen0`, and `sstateen0`:

bit 0 - Custom state

bit 1 - `fcsr` for Zfinx and related extensions (Zdinx, etc.)

Bit 0 controls access to any and all custom state.

(Bit 0 of these registers is not custom state itself; it is a standard field of
a standard CSR, either `mstateen0`, `hstateen0`, or `sstateen0`. The requirements
that non-standard extensions must meet to be _conforming_ are not relaxed due
solely to changes in the value of this bit. In particular, if software sets
this bit but does not execute any custom instructions or access any custom
state, the software must continue to execute as specified by all relevant
RISC-V standards, or the hardware is not standard-conforming.)

Bit 1 applies only for the case when floating-point instructions operate on `x`
registers instead of `f` registers. Whenever `misa`.F = 1, bit 1 of `mstateen0` is
read-only zero (and hence read-only zero in `hstateen0` and `sstateen0` too). For
convenience, when the `stateen` CSRs are implemented and `misa`.F = 0, then if bit
1 of a controlling `stateen0` CSR is zero, _all_ floating-point instructions
cause an illegal instruction trap (or virtual instruction trap, if relevant),
as though they all access `fcsr`, regardless of whether they really do.

In addition to the bits listed above for user-accessible state, the following
are also defined initially for `mstateen0`:

bit 57 - `hcontext`, `scontext`

bits 60:58 - Reserved for the RISC-V Advanced Interrupt Architecture

bit 61 - Reserved for possible `henvcfg2`/`henvcfg2h`, `senvcfg2`

bit 62 - `henvcfg`/`henvcfgh`, `senvcfg`

bit 63 - `hstateen0`/`hstateen0h`, `sstateen0`

The bits defined initially for `hstateen0` are the same as those for `mstateen0`
except applying only to state that is accessible in VS-mode:

bit 57 - `scontext`

bits 60:58 - Reserved for the RISC-V Advanced Interrupt Architecture

bit 61 - Reserved for a possible `senvcfg2`

bit 62 - `senvcfg`

bit 63 - `sstateen0`

(Setting `hstateen0` bit 58 to zero prevents a virtual machine from accessing the
hart's IMSIC the same as setting `hstatus`.VGEIN = 0.)

gfavor, 2024-03-01:
@ved-rivos Could you provide guidance to Bill as to how/where to incorporate the bit name acronyms that JohnH has provided at the beginning of this issue?

Once the bit description text is converted into a register layout diagram, then presumably bit numbers will appear above (or below?) the register diagram showing the acronym name in each field of the register.  And for now, until that happens, what about a format like this  (where ## is the bit number and XYZ is the acronym name):

bit ## (XYZ) - description ...

ved-rivos, 2024-03-06:
== Supervisor State Enable Register (`sstateen0`)

[wavedrom, , ]
....
{reg: [
  {bits:  1, name: 'C'},
  {bits:  1, name: 'FCSR'},
  {bits:  1, name: 'JVT'},
  {bits: 61: name: `WPRI`}
], config:{lanes: 1, hspace:1024}}
....

The `C` bit controls access to any and all custom state.

[NOTE]
===
Bit 0 of these registers is not custom state itself; it is a standard field of
a standard CSR, either `mstateen0`, `hstateen0`, or `sstateen0`. The
requirements that non-standard extensions must meet to be _conforming_ are not
relaxed due solely to changes in the value of this bit. In particular, if
software sets this bit but does not execute any custom instructions or access
any custom state, the software must continue to execute as specified by all
relevant RISC-V standards, or the hardware is not standard-conforming.
===

The `FCSR` bit controls access to `fcsr` for the case when floating-point
instructions operate on `x` registers instead of `f` registers as specified by
the Zfinx and related extensions (Zdinx, etc.). Whenever `misa`.F = 1, bit 1 of
`mstateen0` is read-only zero (and hence read-only zero in `hstateen0` and
`sstateen0` too).  For convenience, when the `stateen` CSRs are implemented and
`misa`.F = 0, then if bit 1 of a controlling `stateen0` CSR is zero, _all_
floating-point instructions cause an illegal instruction trap (or virtual
instruction trap, if relevant), as though they all access `fcsr`, regardless of
whether they really do.

The `JVT` controls access to the `JVT` CSR provided by the Zcmt extension.

== Machine State Enable Register (`mstateen0`)

[wavedrom, , ]
....
{reg: [
  {bits:  1, name: 'C'},
  {bits:  1, name: 'FCSR'},
  {bits:  1, name: 'JVT'},
  {bits: 53: name: `WPRI`}
  {bits:  1: name: `P1P13`}
  {bits:  1: name: `CONTEXT`}
  {bits:  1: name: `IMSIC`}
  {bits:  1: name: `AIA`}
  {bits:  1: name: `CSRIND`}
  {bits:  1: name: `WPRI`}
 {bits:  1: name: `ENVCFG`}
  {bits:  1: name: `SE0`}
], config:{lanes: 1, hspace:1024}}
....

The `C`, `FCSR`, and the `JVT` bits control access to the same state as
controllled by the same bits in `sstateen0` CSR.

The `SE0` bit in `mstateen0` controls access to the `hstateen0`, `hstateen0h`,
and the `sstateen0` CSRs.

The `ENVCFG` bit in `mstateen0` controls access to the `henvcfg`, `henvcfgh`,
and the `senvcfg` CSRs.

The `CSRIND` bit in `mstateen0` controls access to the `siselect`, `sireg*`,
`vsiselect`, and the `vsireg*` CSRs provided by the Sscsrind extensions.

The `IMSIC` bit in `mstateen0` controls access to the IMSIC state, including
CSRs `stopei` and `vstopei`, provided by the Ssaia extension.

The `AIA` bit in `mstateen0` controls access to all state introduced by the
Ssaia extension and is not controlled by either the `CSRIND` or the `IMSIC`
bits.

The `CONTEXT` bit in `mstateen0` controls access to the `scontext` and
`hcontext` CSRs provided by the Sdtrig ISA extension.

The `P1P13` bit in `mstateen0` controls access to the `hedelegh` introduced by
Privileged Specification Version 1.13.

== Hypervisor State Enable Register (`hstateen0`)

[wavedrom, , ]
....
{reg: [
  {bits:  1, name: 'C'},
  {bits:  1, name: 'FCSR'},
  {bits:  1, name: 'JVT'},
  {bits: 54: name: `WPRI`}
  {bits:  1: name: `CONTEXT`}
  {bits:  1: name: `IMSIC`}
  {bits:  1: name: `AIA`}
  {bits:  1: name: `CSRIND`}
  {bits:  1: name: `WPRI`}
  {bits:  1: name: `ENVCFG`}
  {bits:  1: name: `SE0`}
], config:{lanes: 1, hspace:1024}}
....

The `C`, `FCSR`, and the `JVT` bits control access, in VS- and VU-mode, to the
same state as controllled by the same bits in `sstateen0` CSR.

The `SE0` bit in `hstateen0` controls access to the `sstateen0` CSR.

The `ENVCFG` bit in `hstateen0` controls access to the `senvcfg` CSRs.
The `CSRIND` bit in `hstateen0` controls access to the `siselect` and the
`sireg*`, (really `vsiselect` and `vsireg*`) CSRs provided by the
Sscsrind extensions.

The `IMSIC` bit in `hstateen0` controls access to the guest IMSIC state,
including CSRs `stopei` (really `vstopei`), provided by the Ssaia extension.

[NOTE]
===
Setting the `IMSIC` bit in `hstateen0` to zero prevents a virtual machine from
accessing the hart's IMSIC the same as setting `hstatus`.VGEIN = 0.
===

The `AIA` bit in `hstateen0` controls access to all state introduced by the
Ssaia extension and is not controlled by either the `CSRIND` or the `IMSIC`
bits of `hstateen0`.

The `CONTEXT` bit in `hstateen0` controls access to the `scontext` CSR
provided by the Sdtrig ISA extension.

The `CONTEXT` bit in `hstateen0` controls access to the `scontext` CSR.
                                                                                      1

ved-rivos, 2024-03-06:
@wmat - this code replaces lines 206 through 259 in https://github.com/riscv/riscv-isa-manual/pull/1218

@gfavor - please take a look.

wmat, 2024-03-26:
I believe this can be closed as completed.

closed 2024-03-26 by wmat