riscv/riscv-isa-manual #1224: wfi in u-mode when u+s configuration

alienock, 2024-02-12:
Hi, 

I am a little confused about the behavior of wfi in u+s configuration.

As spec in section 3.1.6.5 stated:
"When S-mode is implemented, then executing WFI in U-mode causes an illegal instruction exception, unless it completes within an implementation-specific, bounded time limit. A future revision
of this specification might add a feature that allows S-mode to selectively permit WFI in U-mode.
Such a feature would only be active when TW=0."

My understanding is that, (in u+s) wfi is illegal in u-mode when mstatus.TW=0, while if mstatus.TW=1 it will be legal if completes within an implementation-specific bounded time limit else would be illegal.

Am I interpreting it right or is there something I am missing out.
Please let me know.

Thanks

gfavor, 2024-02-12:
The description of the TW bit (right before your excerpt) says:

_When TW=0, the WFI instruction may execute in lower privilege modes when not prevented for some other reason. When TW=1, then if WFI is executed in any less-privileged mode, and it does not complete within an implementation-specific, bounded time limit, the WFI instruction causes an illegal-instruction exception._

Note that the preceding general statement applies to any and all lower privilege modes that are implemented (which may or may not include S-mode).

Your excerpt then specifies an additional behavior specifically for when S-mode is implemented and only when TW=0 - which falls under the earlier "when not prevented for some other reason" clause.

alienock, 2024-02-13:
So the wfi will be illegal in u-mode when s-mode is implemented with TW=0, while with TW=1 it can be executed.
I think I am understanding it correctly, if not please let me know.

alienock, 2024-10-21:
I was going through the zawrs spec and again got confused by the behavior of wfi in u-mode (s is implemented) by reading the below statement:

"WRS.NTO, unlike WFI, is not specified to cause an illegal instruction exception if
executed in U-mode when the governing TW bit is 0."

When TW=0, would wfi be legal in u-mode (u+s) if it completes within an implementation-specific, bounded time limit OR it will always be illegal (before that feature is implemented)?

I would really appreciate if anyone could clarify me on this.

aswaterman, 2024-10-21:
Implementations are allowed, but not required, to unconditionally trap WFI in U-mode.

closed 2024-10-23 by alienock