riscv/riscv-isa-manual #1234: Privilege mode permission of WFI

YenHaoChen, 2024-02-19:
Hello riscv-isa-manual,

I want to understand the privilege mode permission of WFI and summarize the following table. 

The WFI is allowed in VS-mode when mstatus.TW=hstatus.VTW=0. The WFI may raise illegal instruction in VS-mode if mstatus.TW=1 and may raise virtual instruction in VS-mode if mstatus.TW=0 but hstatus.VTW=1.

I am curious about the privilege mode permission of WFI in VU-mode. Additionally, which exception does the WFI raise on different mstatus.TW and hstatus.VTW values?

```
M-mode    | always allowed
(H)S-mode | allowed if mstatus.TW=0
U-mode    | allowed if mstatus.TW=0 and unimplemented S-mode
VS-mode   | allowed if mstatus.TW=0 and hstatus.VTW=0
VU-mode   | ????????
```

Thank you.

Best regards,
YenHaoChen

YenHaoChen, 2024-02-19:
The spec says that the mstatus.TW affects VU-mode. However, I need help clarifying how the mstatus.TW affects the VU-mode.
[image]

aswaterman, 2024-02-19:
> I need help clarifying how the mstatus.TW affects the VU-mode.

It affects VU-mode in the same way it affects other modes (=1 means no effect, =0 means illegal instruction if doesn't complete in bounded time which may be zero)

YenHaoChen, 2024-02-19:
The spec says the WFI may raise an illegal instruction in U-mode if S-mode is implemented (if it doesn't complete in bounded time). I assume this rule is also applied in VU-mode, meaning that the WFI may raise an exception in VU-mode. I want to understand which exception WFI raises in VU-mode (if doesn't complete in bounded time).

To my intuition, the WFI may raise illegal instruction in VU-mode if mstatus.TW=1 and may raise virtual illegal instruction in VU-mode if mstatus.TW=0 (if doesn't complete in bounded time). Is this correct?

aswaterman, 2024-02-19:
Yeah.

closed 2024-02-19 by YenHaoChen

aswaterman, 2024-02-19:
I re-read the spec, and I think I gave you bad advice.  In the Hypervisor chapter's Trap Cause Codes section, there is a list of situations in which the virtual-instruction exception is raised.  For WFI, it boils down to `mstatus.TW == 0 && (mode == VU || (mode == VS && hstatus.VTW == 1))`.