riscv/riscv-isa-manual #1251: Question about mtval update value when misa.C=0 and compressed instruction is executed

viktoryou, 2024-03-07:
From Priv. 1.12:
> The mtval register can optionally also be used to return the faulting instruction bits on an illegal instruction exception (mepc points to the faulting instruction in memory). If mtval is written with a nonzero value when an illegal-instruction exception occurs, then mtval will contain the shortest of:
> * the actual faulting instruction
> * the first ILEN bits of the faulting instruction
> * the first MXLEN bits of the faulting instruction

In a RVI64 implementation that supports this feature, if the faulting instruction is a 16-bit C-ISA instruction, it is clear mtval should be updated by the actual 16-bit opcode.

When misa.C=0 (not supporting C-ISA by hardware, or dynamically disabling C-ISA), in such implementation, for a faulting 16-bit C-ISA instruction,  should the hardware still report the actual 16-bit opcode? Or, a 32-bit opcode instead is preferred since the instruction is fetched by a 32-bit width implied by misa.C=0?

gfavor, 2024-03-07:
It is the "actual" faulting instruction (not some transformed version of it).

And note that the there is no 32-bit instruction that is actually "fetched".  The only thing fetched is the actual but illegal 16-bit instruction.  And it is the actual 16-bit instruction that is decoded into its 32-bit equivalent or, in this case, recognized as illegal.

viktoryou, 2024-03-07:
Thanks for clarification. More specifically, if the 16-bit instruction is recognized as illegal, about the higher 16-bit in mtval, is it required to be written by all zeroes, or another 16-bit opcode followed is allowed to be written?

gfavor, 2024-03-07:
The mtval arch spec says "The value loaded into mtval on an illegal-instruction exception is right-justified and all unused
upper bits are cleared to zero."

viktoryou, 2024-03-07:
Got it. Thanks again.

closed 2024-03-07 by viktoryou