riscv/riscv-isa-manual #1253: No mention of jr pseudo-instruction

dzaima, 2024-03-09:
The manual mentions `jr` twice - [in the LR/SC example](https://github.com/riscv/riscv-isa-manual/blob/6f6c947dc0b5be01c088b0f335181e586c223a53/src/a-st-ext.adoc?plain=1#L241) and [for `cm.jt`](https://github.com/riscv/riscv-isa-manual/blob/6f6c947dc0b5be01c088b0f335181e586c223a53/src/zc.adoc#table-jump-overview) (and gcc and clang `-S` emit it for indirect calls), but I can't find any description of it in the manual. It should be mentioned as a pseudoinstruction for `jalr x0, 0(rs1)`.

aswaterman, 2024-03-10:
See the RISC-V Assembly Programmer's Manual.  We don't see fit to duplicate that information in the ISA spec.

https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md

abrodkin, 2024-07-17:
> See the RISC-V Assembly Programmer's Manual. We don't see fit to duplicate that information in the ISA spec.
> 
> https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md

I'm wondering why pseudoinstructions don't fit well in the ISA manual. For most of the mere mortals dealing with RISC-V assembly (either reading it or writing) pseudoinstructions don't look any different compared to "normal" instructions. And given pseudoinstructions don't depend on ABI or something else, it would be so much more convenient to just have them listed where the rest of instructions are described. I.e. copy or even move https://github.com/riscv-non-isa/riscv-asm-manual/blob/main/riscv-asm.md#-a-listing-of-standard-risc-v-pseudoinstructions into this ISA manual.

aswaterman, 2024-07-17:
I agree that the current situation, where the ISA spec uses a small number of pseudoinstructions but does not define them, makes it hard on readers.  Solutions include:

- Don't use pseudoinstructions in the ISA manual.  (This is a poor choice, IMO, since the pseudoinstructions are generally easier to read, and since they convey proper style.)
- Have a short section that defines only the pseudoinstructions that the manual uses.  (This avoids duplicating so much material from the ASM manual.)
- Merge the ASM manual into back into the ISA spec at some point in the future.

dzaima, 2024-07-17:
For `jr` in particular, what I found odd is that, just above `jalr`, there is:

> Plain unconditional jumps (assembler pseudoinstruction J) are encoded as a JAL with rd=`x0`.

which does mention a different pseudoinstruction, from which I expected that related pseudoinstructions would be mentioned as they come up.

kasanovic, 2024-07-21:
I remember some other ISA manuals list the pseudo-instructions in the "page-per-instruction" reference section, calling out that these are specialized variants of the other instruction.

aswaterman, 2024-07-24:
We discussed this in the Architecture Review Committee meeting today, and we decided that the ISA spec can normatively define some pseudoinstructions, particularly no-brainers like MV.  It's OK if there's partial overlap with the ASM manual, though the ASM manual is beholden to respecting any ISA-spec pseudoinstruction definitions.

To resolve this specific issue, we will define the JR and RET pseudoinstructions just after the definition of JALR.  As @dzaima points out, this is symmetric with J/JAL.

closed 2024-07-24 by aswaterman

abrodkin, 2024-07-24:
So is my understanding correct that there's no plan for adding all existing pseudoinstructions in the ISA manual, instead we'll act on the case-by-case basis addressing requests for adding particular pseudoinsturctions? Sorry for bringing this up again, but my worry is that now the ISA manual will look inconsistent: some pseudoinstructions are there, while some are not. What's actually holding us from adding all the pseudoinstructions while we're at it?

kasanovic, 2024-07-24:
The ISA manual should list all the standard pseudoinstructions that expand 1:1 to machine instructions, but the ISA manual should not list pseudoinstructions that assembler can expand into more than one machine instruction.  Assemblers may also define additional pseudoinstructions that are 1:1 for some reason, but the ISA manual should only list the canonical ones expected to be supported by all assemblers.

commit ce93bad9 (2024-07-24): Define JR, RET pseudoinstructions

Symmetric with defining J pseudoinstruction earlier in the same section.

Resolves #1253
PR #1555 (2024-07-24): Define JR, RET pseudoinstructions
Symmetric with defining J pseudoinstruction earlier in the same section.

Resolves #1253
commit fb235792 (2024-07-24): Define JR, RET pseudoinstructions (#1555)

Symmetric with defining J pseudoinstruction earlier in the same section.

Resolves #1253