riscv/riscv-isa-manual #1254: Addresses written into mtval, mtval2, stval, htval across XLEN transitions

sorear, 2024-03-11:
We have that U-mode execution "are taken modulo 2^UXLEN", which probably means that traps from U-mode set stval or mtval to a zero extended value in addition to requiring zero extension before address translation.

We do not have any language constraining addresses in HS-mode or VS-mode. Are effective addresses taken modulo 2^HSXLEN or 2^VSXLEN for the purposes of setting stval and mtval? Are shifted guest physical addresses taken modulo 2^VSXLEN for the purposes of setting htval and mtval2? Are guest physical addresses taken modulo 2^(VSXLEN+2) for the purposes of G-stage translation?

gfavor, 2024-03-11:
Yes (for consistency), effective addresses are taken modulo their current XLEN.  With regards to GPAs, they do not exist within a guest.  Their size is a function of HSXLEN (which is spec'ed as such for the CSR that captures GPAs, i.e. htval).

allenjbaum, 2024-03-11:
Should that be zero-extended, or sign extended if MMU is enabled? (and it's
an address vs instruction bits)
I admit to be very confused about handling of virtual machines....

gfavor, 2024-03-11:
In general, VAs architecturally are sign-extended values; PAs (and GPAs) architecturally are zero-extended values.  But note that the arch spec - for mstatus and the *XL fields -  says:

Whenever XLEN in any mode is set to a value less than the widest supported XLEN, all operations must ignore source operand register bits above the configured XLEN, and must sign-extend results to fill the entire widest supported XLEN in the destination register. Similarly, pc bits above XLEN are ignored, and when the pc is written, it is sign-extended to fill the widest supported XLEN.

And the spec in sstatus for UXL with regards to just user-mode addresses and the specific UXLEN/SXLEN combination, overrides that.  Whether that is intentional or an error, I'll let others (especially @aswaterman ) comment.

allenjbaum, 2024-03-12:
I will interpret that as "clarification is needed, and it isn't necessarily
zero-extended".

aswaterman, 2024-03-12:
It's very intentional, as it gives RV32 processes on an RV64 system up to 4 GiB of address space, rather than 2 GiB.  If addresses were instead sign extended, then half of the address space would land in the kernel's space.

sorear, 2024-03-14:
The xtval and xtval2 questions remain open.

aswaterman, 2024-03-14:
I argue the value written to xtval follows from the fact that addresses are taken mod 2^UXLEN.

sorear, 2024-03-15:
The second paragraph of the original post was not about U-mode, so I don't see how "addresses are taken mod 2^UXLEN" helps?

aswaterman, 2024-03-15:
Apologies, @sorear, I was responding in the context of the last few posts in this thread, not to the original post.  But I think that answers the question about stval.

As to the other cases, my reasoning is a bit off-the-cuff, but I think the answer is always to zero-extend.  This is consistent with the CSR width modulation section in the spec.  And it seems to me to be compatible with what software wants.

The reasoning for htval/mtval2 is similar to that for stval: we want the full 34b physical-address space to be available to an RV32 guest in a manner that doesn't create special cases for an RV64 hypervisor, and so zero-extension is the natural choice.  If sign-extension had instead been chosen, then the hypervisor would need to manually clear the MSBs before walking the G-stage page tables--otherwise it would confuse the address with an invalid GPA from an RV64 guest.  And, as @gfavor wrote, zero-extension follows the RISC-V convention that quantities that are notionally physical addresses are unsigned numbers.

What about mtval?  Here, it's more of a don't-care, since most of the time, the mtval address would be used in an MPRV sequence.  The load or store within the MPRV sequence would simply ignore the MSBs if mstatus.MPP corresponded to a mode whose XLEN was less than MXLEN.  And if we don't care, then we might as well do the same thing as for the other cases and zero-extend.

cc @jhauser-us