riscv/riscv-isa-manual #1257: Is a hart a hardware thread?

davidharrishmc, 2024-03-12:
My colleagues and I are struggling to precisely define "hart" and are concerned the spec is misleading.

Section 1.1 paragraph 2 states "A RISC-V compatible core might support multiple RISC-V-compatible hardware threads, or harts, through multithreading."

This language suggests that a hart is a hardware thread.

Section 1.2 states "From the perspective of software running in a given execution environment, a hart is a resource that autonomously fetches and executes RISC-V instructions within that execution environment. In this respect, a hart behaves like a hardware thread resource even if time-multiplexed onto real hardware by the execution environment."

This language suggests that a hart is not a "hardware thread resource" but behaves like one.

Later in Section 1.2 states "The important distinction between a hardware thread (hart)... harts operate like hardware threads."  Part of this paragraph equates harts to hardware threads, and part indicates that they are different but similar ("like").

It would be helpful to define precisely what a hardware thread is, and whether a hart is actually a hardware thread, or if it is an abstraction of a hardware thread.  

To the best of my understanding, a hart is not precisely a hardware thread. I would suggest the following changes to Section 1.

* 1.1 or 1.2 Precisely define hardware thread, and explain how a hart is similar to but not exact the same as a hardware thread.
* 1.1 "might support multiple RISC-V-compatible hardware threads, or harts, through multithreading" -> "might support multiple RISC-V-compatible harts through multithreading"
* 1.2 "distinction between a hardware thread (hart) and a software thread context" -> "distinction between a hart and a software thread context"

gfavor, 2024-03-12:
On Tue, Mar 12, 2024 at 8:28 AM David Harris ***@***.***>
wrote:

> Section 1.1 paragraph 2 states "A RISC-V compatible core might support
> multiple RISC-V-compatible hardware threads, or harts, through
> multithreading."
>
> This language suggests that a hart is a hardware thread.
>
Yes, this strongly "suggests" that  "hart" = "hardware thread".

> Section 1.2 states "From the perspective of software running in a given
> execution environment, a hart is a resource that autonomously fetches and
> executes RISC-V instructions within that execution [...]
This text and the surrounding paragraph is trying to say that the
environment’s harts operate like hardware threads - versus like software
threads - from the perspective of the software inside
the execution environment.  In other words the key point being made isn't
about hart versus hardware thread, but about software thread versus
hardware thread.

> It would be helpful to define precisely what a hardware thread is, and
> whether a hart is actually a hardware thread, or if it is an abstraction of
> a hardware thread.
>
> To the best of my understanding, a hart is not precisely a hardware
> thread. I would suggest the following changes to Section 1.
>
>    - 1.1 or 1.2 Precisely define hardware thread, and explain how a hart
>    is similar to but not exact the same as a hardware thread.
>
> 1.1 (as excerpted above) clearly defines "hart" = "hardware thread".

>
>    - 1.1 "might support multiple RISC-V-compatible hardware threads, or
>    harts, through multithreading" -> "might support multiple RISC-V-compatible
>    harts through multithreading"
>
>  This seems fine (and it reinforces explicitly that "hardware thread" =
"hart").

>
>    - 1.2 "distinction between a hardware thread (hart) and a software
>    thread context" -> "distinction between a hart and a software thread
>    context"
>
> The former seems fine and better in that it makes clear that  "hardware
thread" = "hart".

> Message ID: ***@***.***>
>

nick-knight, 2024-03-12:
My understanding is that a "hart" is a fundamental notion in this abstract computation model, and the authors are mentioning hardware threads only by analogy, to illustrate how this abstract notion may be implemented in real hardware. Please correct me if this understanding is wrong.

gfavor, 2024-03-12:
I would say that the spec repeatedly and directly affirms that "hart" = "hardware thread", i.e. that doesn't seem to be a thing that is ambiguously suggested once in the spec and never mentioned or stated again.

Having said that, some would view "hardware thread" as an abstraction.  (While others would view that as a physical reality and "software thread" is an abstraction, but then again to software people that is their reality.)

In short, the spec says that "hart" = "hardware thread" and reaffirms that a few times.  From some perspectives that may be an abstraction (e.g. from a multi-threaded CPU perspective), and other perspectives that is a concrete physical idea.  Neither is wrong.  But from the ISA specs' perspective, yes, "hart" is a form of abstraction - that is grounded as a hardware level thread of execution and not as a software thread (as OS people would think in terms of).

kasanovic, 2024-03-12:
Yes, a "hart" is an abstraction of a hardware thread.  So there are many possible concrete implementations of a hart, include a real hardware thread, a QEMU emulation of a hardware thread, etc.

kasanovic, 2024-03-12:
@pdonahue-ventana A hart is a thing that independently runs forward, continually executing instructions inside some execution environment.  The text you quoted from spec is saying that while each user-level execution environment sees one or more harts running code at user-level, in reality under the hood, the operating system is sharing physical processor cores among the user-level execution environments.  The physical cores only advance the state of each user-level execution context when a core is running a hart from that context.  The hart abstraction is needed so we don't have to discuss physical cores when they're running a different context, while also preserving the concept of something inside each context that has register state and that advances independently, executing instructions from the current context.

davidharrishmc, 2024-03-13:
Thank you.  This is helpful to define a hart as "an abstraction of a hardware thread."

If I understand this correctly, a this abstraction means that hart could be a physical hardware thread, which I believe comprises its own architectural state (PC, registers, and CSRs), and the ability to fetch and execute instructions. It could also be a QEMU emulation of a hardware thread.  It could also be a guest hart, which is time-multiplexed onto a a host hart. Examples of guest harts are user-level harts managed by host hart(s) running a supervisor-mode operating system, and guest OS harts managed by host hart(s) running a hypervisor.

Page 4 uses the terms "guest hart" and "host hart".  These terms don't appear elsewhere.  Are they the preferred language for referring to time-multiplexing harts?  The term "virtual hart" comes to me for harts time-multiplexed onto other harts, by analogy to virtual memory and virtual machines, but it isn't in the spec.

On page 4, should I read "The distinction between a hardware thread (hart) and a software thread _context_" as ["hardware thread _context_" vs. "software thread _context_"] or ["hardware thread" vs. "software thread _context_"]?  The language is ambiguous.  The term "context" here is also undefined.  Does it mean the values of the architectural state (PC and registers, as in a context switch, as used in Privileged Spec 3.1.6.6), or does it relate to the execution environment (the context in which the thread is running, as seems to be used in Pan's thesis "Cooperative Hierarchical Resource Management for Efficient Composition of Parallel Software" at page 32), or something else?

Just beyond in page 4, I see that a hart differs from a software thread (or software thread context) in that software running inside the execution environment is not responsible for causing progress of each of its harts.  I get that - I used to program MacOS in the 1980s when it was cooperative multitasking, and a process that failed to yield would tie up the whole machine.  However, a modern preemptive multitasking operating system doesn't have that problem.  In preemptive multitasking, does every Pthread run on one user-level hart, which might be a guest hart time-multiplexed onto a physical hart?  Thus, is there a one-to-one correspondence between software threads and user harts in a preemptive multitasking system?

@kasanovic you use the term "Execution Context" (EC) in your reply above, which is not in the spec except one mention in 3.6 "PMAs do not vary by execution context".  What is your definition of EC? Should this term be defined in the spec before being used in 3.6, and used elsewhere if it is helpful to explain harts?

Sorry for so many questions - I want to be sure I'm using the terms precisely because they are not standardized across the literature.

kasanovic, 2024-03-14:
I use execution context in the industry-standard way to refer to the state of a computation (as used in the phrase "context switching").  An execution environment enables the state of an execution context to advance (so a simple hardware microcontroller provides a low-level M-mode execution environment which advances the state of the single M-mode execution context by executing instructions and taking interrupts in hardware, while a hypervisor provides multiple supervisor-level execution environments which each can advance the state of a guest-OS execution context).  The interface between an execution context and an execution environment is the "execution environment interface (EEI)", which is an abstract interface that defines the available memory spaces, the instructions supported, behavior of requested traps, and any other per EEI behavior.  The Linux ABI provides an EEI for user processes, and includes things like how the execution context is initialized, what instructions can be executed, what memory can be touched, what syscalls are supported and their effect on the execution context (including syscalls that add harts to the context) and how to interact with the outside world, etc.  An abstract EEI can have many different concrete execution environment implementations (e.g., an actual Linux  kernel running on hardware or a QEMU user-mode emulation can both provide a user-level EE obeying the Linux ABI which can advance the state of a user-level execution context). Note that it is possible to migrate an execution context from one concrete execution environment to another, as is done when migrating VMs around a datacenter or when snapshotting simulations to move between functional-only and cycle-accurate simulations.

davidharrishmc, 2024-03-14:
I think of context switching as saving and restoring the PC, registers, and CSRs.  That's a lighter weight operation than migrating VMs around a data center. Does "execution context" refer to {PC, registers, CSRs} or does it also include the memory used by a hart (and perhaps state of certain I/O devices), that would be necessary to migrate the hart around a data center?

Returning to the previous question, is there a one-to-one correspondence between software threads and user harts in a preemptive multitasking system?

kasanovic, 2024-03-14:
"context" is a generic term.  When a unix-style OS switches a physical core from one user process to another, the VM page table pointer is switched to present a completely different view of user memory. The only actual physical data movement is swapping the registers in this case, but conceptually the physical core moved into an entirely different memory context.  Migrating a VM around a data center involves more physical copying of data, but is the same concept (moving a context on/off different physical resources). 

A user hart is an abstraction of a processor running instruction in that user context.  The problem with equating this to a preemptive software thread is that software threads have their own specifications as to operations they perform and how they synchronize depending on the threading model.  If we take the most primitive common software thread, a pthread, then that can be implemented by a hart, but pthreads provide far more sophisticated synchronization operations that rely on the execution environment interface.  I don't believe you can implement all pthread operations using only harts with no support from the execution environment.  A hart is just something that fetches and executes instructions within an execution context.

kersten1, 2024-04-11:
Just to pile on, this is the definition of Hart that is in the glossary that I'm resurrecting. 
----
HART
Hardware Thread. At machine-mode level, each hart is a real hardware thread, either one hart per core without hardware multithreading, or multiple harts per core with hardware multithreading, and 'hart' represents the hardware resource. It is possible to emulate harts in software; for example, privileged execution environments can multiplex lesser-privileged harts onto physical hardware using timer interrupts. Note that co-operative multithreading within the same privilege level is not a compliant implementation, however. Across all implementation choices, the concept of a hart is defined as a resource abstraction representing an independently advancing RISC-V execution context within a RISC-V execution environment.
-----

Is this an acceptable definition? My hope is that some day we can just link to the glossary for definitions, making them all single source. 

Glossary - https://github.com/riscv/riscv-glossary

kasanovic, 2024-06-03:
@kersten1 A hart is an abstraction of a hardware thread, meaning that it captures the important aspects of a real hardware thread for the purposes of defining the RISC-V specifications.  So we should not equate hart with a "real hardware thread" - a hart is anything that has the appearance of executing instructions within an execution context.  All harts are considered equal from the perspective of the specifications, whether real or emulated.

kersten1, 2024-06-06:
Hart
An abstraction of a hardware thread that has the appearance of running instructions within an execution context. A hart captures the important aspects of a real hardware thread that are used to define the RISC-V specifications.  All harts are considered equal from the perspective of the specifications, whether real or emulated.

@kasanovic Sorry to keep poking on this but I feel that this is an important concept to get right.... :)

davidharrishmc, 2024-06-16:
n the unpriv-isa-asciidoc_20240411, page 554 or page 563 defines R-type and I-type formats, which differ only in that instr[31:20] = {funct7, rs2} for R-type and imm12 for I-type.

Some instructions such as fsqrt, fcvt, cpop, and zip use a single register, and contain constants in the instr[24:20] that serve as an extended function code rather than an immediate.  It appears that the unprivileged spec is silent about their type. Is there a collective understanding of type for these instructions?

Is it best to classify these instructions as I-type even though the immediate is not used as a source operand?  Or better to call them R-type with rs2 hardwired to a funct5 value?

Sorry this question is pendantic, but I’m a pedant.

Thank you,

David

aswaterman, 2024-06-17:
The question is totally off-topic given the thread is about the definition of hart. But I’ll briefly comment that I-type would also have been a poor classification because there’s no immediate operand. We use instruction types as classification tools, not as strictures. We’d have hundreds of distinct types if we used that tool with precision—-and at that point, it wouldn’t be doing anyone any service.

davidharrishmc, 2024-06-17:
My error replying to an old post.  I'll start a new issue.

kasanovic, 2024-06-18:
@kersten1  How about:

Hart: An abstraction of a hardware thread that captures the important aspects of a real hardware thread for the purposes of defining the RISC-V specifications.  In particular, a hart is the agent that executes instruction within an execution context.

gfavor, 2024-06-18:
"instruction" -> "instructions".

And maybe "RISC-V ISA specifications".

kasanovic, 2024-06-18:
@gfavor The term hart is used in non-ISA specifications, but yes, should have plural instructions:

Hart: An abstraction of a hardware thread that captures the important aspects of a real hardware thread for the purposes of defining the RISC-V specifications. In particular, a hart is the agent that executes instructions within an execution context.

closed 2025-12-09 by aswaterman