riscv/riscv-isa-manual #1258: May I ask why the two directives HLVX.HU and HLVX.WU are added? Are there any special uses for them?

GZYyyyy, 2024-03-13:
The privilege document has the following description: 
_Instructions HLVX.HU and HLVX.WU are the same as HLV.HU and HLV.WU, except that execute
permission takes the place of read permission during address translation. That is, the memory being
read must be executable in both stages of address translation, but read permission is not required.
For the supervisor physical address that results from address translation, the supervisor physical
memory attributes must grant both execute and read permissions. (The supervisor physical memory
attributes are the machine’s physical memory attributes as modified by physical memory protection,
Section 3.7, for supervisor level.)_

These two instructions allow LSU to obtain instructions. What is the use after obtaining the instructions? Is there any special use of the software? Can you describe a typical scenario in which these two instructions are used in software?

gfavor, 2024-03-13:
HLVX is used by a hypervisor to read guest code (hence the checking of execute permission instead of read permission), e.g. for emulating an instruction that was trapped to the hypervisor.

GZYyyyy, 2024-03-14:
OK, Thanks.

closed 2024-03-14 by GZYyyyy