riscv/riscv-isa-manual #1259: Where does the CBO/CMO specification live?

Timmmm, 2024-03-13:
[This repo](https://github.com/riscvarchive/riscv-CMOs/) has been archived, and the content doesn't seem to have been moved into this repo. Where am I supposed to report/fix issues?

In particular [this ambiguity](https://github.com/riscvarchive/riscv-CMOs/issues/58) needs to be written into the actual spec, rather than spec-by-github-comment. Also the value to write to `mtval` for exceptions needs to be specified.

wmat, 2024-03-13:
Please submit your recommended change as a PR against the riscv-isa-manual CMO chapter, thanks.  Add a reference back to this issue in the original repo.

Timmmm, 2024-03-15:
Ah ha that's embarrassing - I don't know why I couldn't find that chapter when I searched! Thanks!

closed 2024-03-15 by Timmmm