riscv/riscv-isa-manual #1261: mie.MEIE missing inf figure 16

tzwaenn, 2024-03-14:
Looking for mie.MEIE in figure 16, we see that it is missing, screwing up the layout and adding to a trailing "2" for the last 2 bits.
[image]

tzwaenn, 2024-03-15:
The LCOFI bits are a new addition, so they should stay in IMHO.

wmat, 2024-03-15:
How does this look?

[image]

tzwaenn, 2024-03-15:
13 is LCOFIE, 11 MEIE (same as in fig 15 mip just with trailing p instead of e)

wmat, 2024-03-15:
Am I missing a MEIE bit in there?  Fig 15 has a MEIP at bit 11.

wmat, 2024-03-15:
NVM, I see that's what you said.

wmat, 2024-03-15:
How's this now?
[image]

wmat, 2024-03-15:
Fix pushed.

closed 2024-03-15 by wmat