riscv/riscv-isa-manual #1269: Maybe wrong fr edge in litmus sample?

GTwhy, 2024-03-19:
In the provided [litmus test sample](https://github.com/riscv/riscv-isa-manual/blob/riscv-isa-release-157641b-2024-03-12/src/images/graphviz/litmus_sample.png), the edge `c -fr-> a` indicates that event `c` precedes `a`. However, `a` is not a co-successor of `c`, and `c` reads the value `1` from `a` within this example.

Could this be a mistake, or is there a logical explanation for this apparent inconsistency?

Thanks! @wmat 

[image]

wmat, 2024-03-19:
@Andrew Waterman ***@***.***> can you comment here?  This litmus
test image is copied directly from the published version of the spec.

aswaterman, 2024-03-19:
I'll leave it to the memory-model folks lurking on this list to explain the conventions here.

aswaterman, 2025-12-09:
@daniellustig can you take a look or tag someone else?

daniellustig, 2026-01-02:
I think this may just be a confusing artifact of the tooling that originally drew these kinds of graphs.

Per the discussion that follows the cited text, there's already a contradiction in assuming `a->b` and `b->a` simultaneously.  The `fr` edge is just a weird artifact of automated modeling trying to further reason about the implications of that contradiction and draw a picture of that.

(Specifically: if the model tries to satisfy `a0==1` by assuming both `a->b` and `b->a`, then it sees `a--rf-->c` and `a--co-->b--co-->a`.  Coherence order is transitive, so the latter implies `a--co-->a`, and together that implies `c--fr-->a`.  I think that's where this `c--fr-->a` edge came from.)

But more broadly, this figure might be more confusing than helpful.  It might benefit from being redrawn or split into two cases or something.

aswaterman, 2026-01-02:
Thanks for the explanation, @daniellustig.  @GTwhy does that answer your original question?  If so, I'm content to close this issue.