riscv/riscv-isa-manual #1280: MMWP name violates friendly language policy

wmat, 2024-03-21:
MMWP "Machine Mode Whitelist Policy (mseccfg.MMWP)" violates the organizational friendly terminology policy.

As suggested by @sorear, renaming MMWP to MMDD (Machine Mode Default Deny) to make the meaning of 0 and 1 more immediately obvious, and also to comply with the RISC-V Friendly Terminology Policy.

ved-rivos, 2024-03-21:
It was unfortunate it was ratified that way but changing the ratified specification in this manner may not be such a good idea.

wmat, 2024-03-21:
Agreed.  I'll leave this here for further discussion.  Similarly, an alternative name could be "Machine Mode Allowlist Policy (mseccfg.MMAP)" which seems closer to the original intent.

aswaterman, 2024-03-21:
Agreed that changing the acronym is more problematic than changing the name, since the acronym shows up in more places, e.g. header files.

aswaterman, 2024-03-23:
Another option is for W to stand for alloWlist. It’s perhaps silly to use an acronym rather than an initialism, but it’s less of a mouthful than the proposed alternative. In any case, I’d like one of the spec authors to make the call.

kasanovic, 2024-03-25:
Andrew's suggested acronym expansion is fine with me.

wmat, 2024-03-25:
OK, I've updated to PR.

kersten1, 2024-05-07:
This is complete so closing.

closed 2024-05-07 by kersten1