riscv/riscv-isa-manual #1282: more chapter re-ordering

aamartin0000, 2024-03-22:
Thank you for rearranging the chapters to a more sane ordering.  Some suggestions, take 'em or leave 'em:

- in Ch28, Zbkb should be before Zbkc.  Now that the Zbk* descriptions are within the bitmanip chapter, would it be worth mentioning overlaps (Zbkb in Zbb, Zbkc == Zbc)?
- Ch33 (scalar crypto) should be before Ch32 (vector crypto)
- please add sample code for use of (scalar and vector) crypto instructions

kersten1, 2024-03-29:
@aswaterman I can do this if you agree with it.

aswaterman, 2024-03-29:
I think some of the reorg might’ve already been done, but the rest makes sense, so go for it. The sample code is a longer-term thing we don’t need to worry about now.

aamartin0000, 2024-03-29:
Agreed.  Please close this issue when the pull request is opened.

Perhaps members of the crypto SIG/TG can contribute sample (scalar, vector) crypto code when the time is right.

wmat, 2024-04-05:
Closing as reordering patch applied.

closed 2024-04-05 by wmat