riscv/riscv-isa-manual #1286: hlvs

zhangjingru111, 2024-03-25:


zhangjingru111, 2024-03-25:
In spec, hlv and hlvx is transformed with rs2 field, but hlvx doesn't have rs2. Should rs2 fields write 0 or 0x00011 in mtinst when hlvx instrution happened guest fault?
[image]

zhangjingru111, 2024-03-25:
[image]

jrahmeh, 2024-09-16:
I believe for hlvx.hu/hlvx.wu, the rs2 field must be preserved since it is necessary for the correct decoding of those instruction. So in the transformed instruction corresponding to halve.hu/hlvx.wu, the rs2 field should be binary 00011.

aswaterman, 2024-09-16:
Yeah, as the caption says, "all fields are the same as the trapping instruction" except [...], meaning that bits 24:20 are preserved, even though they don't represent rs2.

zhangjingru111, 2024-09-18:
Ok, thank you very much:)

closed 2024-09-18 by zhangjingru111