riscv/riscv-isa-manual #1297: Userspace unique CPU identifier

wangpc-pp, 2024-03-27:
When optimizing `__vdso_getcpu`, we need an userspace unique identifier as the CPU id. Currently we just invoke the syscall.
```
#include <linux/linkage.h>
#include <asm/unistd.h>

	.text
/* int __vdso_getcpu(unsigned *cpu, unsigned *node, void *unused); */
SYM_FUNC_START(__vdso_getcpu)
	.cfi_startproc
	/* For now, just do the syscall. */
	li a7, __NR_getcpu
	ecall
	ret
	.cfi_endproc
SYM_FUNC_END(__vdso_getcpu)
```
But we can't read `mhartid`, is there a way/extension to expose `mhartid` to userspace? Or is there an unique identifier that can be used as CPU id?

wangpc-pp, 2024-04-09:
Ping for comments.

aswaterman, 2024-04-09:
No, there is no plan to add such a feature.

wangpc-pp, 2024-04-09:
> No, there is no plan to add such a feature.

If so, then there is no meaning for `getcpu` in vDSO, which may cause some performance issues compared to other archs.
What if we add a extension to expose `mhartid` to unprivileged code?

aswaterman, 2024-04-09:
There _is_ a reason to have `getcpu` in the vDSO, which is that a user-space facility could eventually be added if deemed necessary.  Perhaps its day will come.

Directly exposing `mhartid` is a virtualization hole.