riscv/riscv-isa-manual #1302: W[13]

nibrunieAtSi5, 2024-03-28:
I noted a small discrepancy in `sha2ms` documentation.

The I/O table leaves the elements between `W[12]` and `W[14]` in `vs1` blank.
[image]

but the SAIL code names this element:
```
    {W[15] @ W[14] @ W[13] @ W[12]} : bits(EGW) = get_velem(vs1, EGW, i);
```

The input word `W[13]` is never actually used, but the I/O table tends to indicate that the instruction does not expect any specific value in its position while the SAIL code seems to indicate the instruction does.

I think this could be clarified.

aswaterman, 2024-03-28:
Is this a @kdockser question?

kdockser, 2024-05-30:
While the pseudo code disagrees with the arguments table on the naming of W[13], I think that it is reasonably clear that there is no specific value required here. In practice, that location will hold W[13], but it will not be used as indicated by the pseudo code. I also think that the point that W[13] is not needed is reinforced with the text:
```
Eleven of the last 16 SEW-sized message-schedule words from vd (oldest), vs2, and vs1 (most recent)
are processed to produce the next 4 message-schedule words.
```
I don't think that further clarification is needed.

closed 2024-05-30 by kdockser