riscv/riscv-isa-manual #1310: Is OP-P the official name of major opcode 0b1110111?

topperc, 2024-04-01:
The Zvk* extensions call it OP-P in their encoding tables. The opcode table in the `RV32/64G Instruction Set Listings` calls it "reserved".

CC @aswaterman

aswaterman, 2024-04-02:
For context, the current P-extension proposal moves those instructions elsewhere, but OP-P is used for Zvk* etc., so another name might be more appropriate.

aswaterman, 2024-04-02:
@topperc The ARC wanted to know whether renaming the major opcode was a headache for toolchains.  If it's not onerous to change, we would like to rename it to OP-VE ("Vector Extensions") to more accurately reflect its current use and likely future use.

topperc, 2024-04-02:
> @topperc The ARC wanted to know whether renaming the major opcode was a headache for toolchains. If it's not onerous to change, we would like to rename it to OP-VE ("Vector Extensions") to more accurately reflect its current use and likely future use.

The only place I can think the name might be exposed to users by the toolchain is in the `.insn` handling. It doesn't look like binutils documented or implemented a name https://sourceware.org/binutils/docs/as/RISC_002dV_002dFormats.html. I think LLVM did expose it as OP_P. LLVM doesn't have its own documentation for .insn so if someone figured that out they had to look at the source code. So the risk is low.

Within the toolchain source code it should be a quick search and replace.

I think we should proceed with renaming.

aswaterman, 2024-04-03:
Sounds reasonable to me, especially if we change it promptly.

PR #1311 (2024-04-05): Opcode 0x77 is named OP-VE, for "Vector Extensions"
Resolves #1310

This major opcode is currently used by vector crypto extensions.  It'll likely be used by future vector extensions, as well.

The official naming table currently describes this opcode as _reserved_, which is obviously no longer the case.  Informally, it has sometimes been referred to as OP-P, since there was once a plan to encode the packed-SIMD extensions there.  That plan has been superseded by a proposal to use brownfield space elsewhere.

The vector crypto spec reflects the informal usage of OP-P; this PR fixes that up, to [... cut]
commit 2a0a7c44 (2024-04-05): Opcode 0x77 is named OP-VE, for "Vector Extensions"

Resolves #1310

This major opcode is currently used by vector crypto extensions.  It'll
likely be used by future vector extensions, as well.

The official naming table currently describes this opcode as _reserved_,
which is obviously no longer the case.  Informally, it has sometimes been
referred to as OP-P, since there was once a plan to encode the packed-SIMD
extensions there.  The vector crypto spec reflects this informal usage;
this commit fixes that up.  (The plan to use OP-P for packed SIMD has been
superseded by a proposal to use brow [... cut]
closed 2024-04-05 by aswaterman