riscv/riscv-isa-manual #1312: [Sstc] Are writes to read-only henvcfg.STCE ignored when menvcfg.STCE=0?

YenHaoChen, 2024-04-03:
Hello,

I am curious whether writes to the read-only-zero henvcfg.STCE are ignored when menvcfg.STCE=0.

The preliminary isa spec says, "writes to read-only bits (of a RW CSR) are ignored." Does the rule apply to the read-only henvcfg.STCE when menvcfg.STCE=0?

[image]

@jscheid-ventana

gfavor, 2024-04-03:
The sstc spec says "When STCE in menvcfg is zero, ..., STCE in henvcfg is read-only zero."  So standard read-only-zero behavior applies, i.e. this bit is not an exception to the general rule (since there is no explicit specifiation of different behavior).

YenHaoChen, 2024-04-03:
Thank you. Thus, when mevncfg.STCE=0, henvcfg.STCE is read-only zero, and writes to the read-only-zero henvcfg.STCE are ignored.

closed 2024-04-03 by YenHaoChen

viktoryou, 2024-04-18:
In a previous issue: https://github.com/riscv-software-src/riscv-isa-sim/issues/1497, we came to the conclusion that in such situation, from the Spec, the bit of henvcfg(pbmte, stce, adue) doesn't require a determinism. Since the spike finally changes into ignoring the write, should this issue be regarded as a bug or just a preference for implementation? If there is the bug, why the explanation in section 2.4 of the Privileged Spec doesn't work here?

YenHaoChen, 2024-04-19:
Both statements apply to this case, although software may not easily observe the violation.

+ (Section 2.1) ... writes to read-only bits (of a RW CSR) are ignored.
+ (Section 2.4) ... change from previously read-only to now writable, that field immediately gets an unspecified but legal value ...

The statements require an implementation to ignore writes to read-only bits, but they allow an unspecified legal value in the bits when they change from read-only to writable. The requirement implies no side effect on writing read-only bits, and the allowing clarifies that changing writability may also change the value.

rbuchner-aril, 2024-05-14:
Do you know where in the spec I can find this: `The spec requires the value function to be independent of the writes when the bits are read-only.`?

YenHaoChen, 2024-05-16:
The statements are the rewordings of:
+ ```(Section 2.1) ... writes to read-only bits (of a RW CSR) are ignored.```
+ ```(Section 2.4) ... change from previously read-only to now writable, that field immediately gets an unspecified but legal value ...```

Ignoring writes implies no side effects. In other words, the ignored writes do not contribute to the value by any means. Thus, the value function is independent of the ignored writes.

rbuchner-aril, 2024-05-21:
Thank you for clarifying that for me!