riscv/riscv-isa-manual #1314: `Zbc` missing from `B` extension description?

TommyMurphyTM1234, 2024-04-03:
From the latest draft:

* https://github.com/riscv/riscv-isa-manual/releases/tag/riscv-isa-release-af07c84-2024-03-26

Shouldn't `Zbc` be mentioned here in the part highlighted in red?
Especially given that it's mentioned in the parts highlighted in orange?

[image]

ved-rivos, 2024-04-03:
B only implies Zba + Zbb + Zbs.

TommyMurphyTM1234, 2024-04-03:
Thanks @ved-rivos. So why is the `Zbc` extension described under the chapter heading _"B" Standard Extension for Bit Manipulation, Version 1.0.0_ if it seemingly has nothing to do with `B`? Seems confusing?

TommyMurphyTM1234, 2024-04-03:
So, the bit-manpulation extension "hierarchy" is basically:

- (All) bit-manipulation extensions
  - B
    - Zba
    - Zbb
    - Zbs
  - Zbc

Does `B` also mean that `b` in an architecture string implies `zba_zbb_zbs` (e.g. `rv64imafdcb/lp64d == rv64imafdc_zbb_zbc_zbs/lp64d`?)
Should that work with the latest `riscv-gnu-toolchain` based on GCC 13.2?
Or only with the latest upstream GCC master or something else?

ved-rivos, 2024-04-03:
That looks right. The [patches](https://www.mail-archive.com/gcc-patches@gcc.gnu.org/msg333693.html) have been created for GCC. I am not sure which version of toolchain.

TommyMurphyTM1234, 2024-04-04:
> That looks right. The [patches](https://www.mail-archive.com/gcc-patches@gcc.gnu.org/msg333693.html) have been created for GCC. I am not sure which version of toolchain.

Thanks @ved-rivos. This post:

* https://www.mail-archive.com/gcc-patches@gcc.gnu.org/msg334092.html

seems to suggest that the patch has still not been integrated but it's not clear to me what "proposal" is holding it up? Maybe a proposal to clarify that `B/b` means `Zba/Zbb/Zbs` and not also `Zbc`?

Compiling the RISC-V toolchain (`riscg-gnu-toolchain`) also shows that the changes in that patch are not present in the upstream GCC master.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1449#issuecomment-2035088804

ved-rivos, 2024-04-04:
> seems to suggest that the patch has still not been integrated but it's not clear to me what "proposal" is holding it up?

Perhaps the maintainers were waiting for the specification to Freeze. The specification has Frozen and completed public review since then and is in process of being ratified.

> Maybe a proposal to clarify that B/b means Zba/Zbb/Zbs and not also Zbc?

That was the done and is the basis for the patch : https://github.com/riscv/riscv-b

TommyMurphyTM1234, 2024-04-04:
Thanks @ved-rivos.

> Perhaps the maintainers were waiting for the specification to Freeze. The specification has Frozen and completed public review since then and is in process of being ratified.

That's confusing - according to this it was ratified November 2021:

* https://wiki.riscv.org/display/HOME/Ratified+Extensions#:~:text=RISC%2DV%20Bit,Zbb%2C%20Zbc%2C%20Zbs

The date of the patch mailing list post is 13 February 2024.

ved-rivos, 2024-04-04:
I hope you saw : https://github.com/riscv/riscv-b

While the Zba, Zbb, Zbs, and Zbc extensions were ratified in Nov 21, the official standardization of B as representing the collection of the Zba, Zbb, and Zbs extensions was done recently and is the point of the patch. The official B extension is now Frozen and has completed public review and expected to be ratified in a few weeks.

TommyMurphyTM1234, 2024-04-04:
> I hope you saw : https://github.com/riscv/riscv-b
>
> While the Zba, Zbb, Zbs, and Zbc extensions were ratified in Nov 21, the official standardization of B as representing the collection of the Zba, [...]

Sorry - I didn't. Thanks for pointing that out. 👍

ved-rivos, 2024-04-04:
The ratification of the B standard extension i.e. codifying B=Zba+Zbb+Zbs is not yet done. It is expected to be ratified by end of this month. Thats why it does not show up under ratified extensions yet.

If you navigate from https://wiki.riscv.org/display/HOME/Specification+Status to 
https://wiki.riscv.org/display/HOME/All+Specifications+Under+Development
You will see
"[B Extension Components](https://jira.riscv.org/browse/RVS-2006?src=confmacro)	SPECIFICATION IN RATIFICATION-READY	ISA	Yes	https://github.com/riscv/riscv-b	Zba,Zbb,Zbs	Apr 25, 2024	Apr 04, 2024"

TommyMurphyTM1234, 2024-04-04:
> If you navigate from https://wiki.riscv.org/display/HOME/Specification+Status to https://wiki.riscv.org/display/HOME/All+Specifications+Under+Development You will see "[B Extension [...]

Thanks @ved-rivos - I had gone to that page but simply did browser page search for "bit" and "manip" and didn't see anything so I didn't notice that one.

ved-rivos, 2024-05-10:
@TommyMurphyTM1234 - The ratified extension is published [here](https://wiki.riscv.org/display/HOME/Ratified+Extensions) .

aswaterman, 2025-12-09:
Closing, as the ratified spec removes all doubt.

closed 2025-12-09 by aswaterman