riscv/riscv-isa-manual #1321: The description of zext.w rd, rs is "Add unsigned word"

eak, 2024-04-04:
The description of zext.w rd, rs is "Add unsigned word" rather than "Zero-extend word" zext.w is a pseudoinstruction for add.uw, so this is a weird situation. It would be good if psuedoinstruction status was indicated in the table, and I still think the description should be Zero-extend word.

nick-knight, 2024-04-04:
IIRC, there was an effort to move things concerning pseudoinstructions out of the ISA manual and into the Assembly Manual, here:

https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#-a-listing-of-standard-risc-v-pseudoinstructions

For example, in that table, it says:

Pseudoinstruction            | Base Instruction(s)                                           | Meaning   | Comment
:----------------------------|:--------------------------------------------------------------|:----------|:--------|
zext.w rd, rs                | slli rd, rs, XLEN - 32; srli rd, rd, XLEN - 32                | Zero extend word | It will expand to another instruction sequence when B extension is available*[1]

I'm still a proponent of having the Assembly Manual be the canonical reference for pseudoinstructions.

aswaterman, 2024-04-04:
> I'm still a proponent of having the Assembly Manual be the canonical reference for pseudoinstructions.

Yes, the ISA spec will still mention certain stylized uses of instructions like this, especially if they might affect microarchitectural decisions.  But the assembly manual should be the authority.

eak, 2024-04-05:
I have no objection to the ASM manual being the master (there is not currently a list of pseudoinstructions in the ISA manual anyway, just scattered references), but it would be good to change the zext.w description in the ISA manual to "zero-extend halfword (pseudoinstruction for add.uw)".  Of course it would also be good to update the ASM manual, as the text Nick cited is out of date I believe (the assembler presumably having a new expansion depending on the ISA string?). It might also be good to include a pointer in the ISA manual to the ASM manual (e.g. in the opcode listing, so someone doesn't wonder why certain opcodes don't exist). I don't know how cross-manual references are supposed to work however.

aswaterman, 2024-04-05:
You're obviously right, @eak.  It took me a while to find the bug in the manual because it's the result of a cross-listing: the text "Add unsigned word" comes from a cross-link to `add.uw` and is not directly included in the table.  I'm unilaterally deciding that the right thing to do is to remove `zext.w` from the table altogether.

aswaterman, 2024-04-05:
Resolved by 7cbf1dcf6ea84958b29a1ce60e9c95a349feb7e7

closed 2024-04-05 by aswaterman

KrystalDelusion, 2024-10-10:
@aswaterman this has reverted and 'zext.w' is appearing in the documentation again.

aswaterman, 2024-10-10:
Fixed again in #1676