riscv/riscv-isa-manual #1340: Some state-enable should be normative.

jhauser-us, 2024-04-10:
In the chapter on extensions Smstateen/Ssstateen, the following text is incorrectly formatted as commentary:

> The FCSR bit controls access to fcsr .... regardless of whether they really do.

and:

> The AIA bit in hstateen0 controls access to all state introduced by the Ssaia extension and is not controlled by either the CSRIND or the IMSIC bits of hstateen0.

These statements should be normative.  (Else, there is no normative specification for bits FCSR and AIA.)

aswaterman, 2024-04-10:
@ved-rivos since you integrated that spec, can you address this one?

ved-rivos, 2024-04-10:
I didn't integrate this chapter but I can address this one. Should I fix up other sections in the chapter as well as part of this -  such as renaming the "Proposal" section and removing duplicated  description of mstateen ?

aswaterman, 2024-04-10:
Oh, my mistake.  Yeah, if you don't mind doing that, it would be much appreciated.

ved-rivos, 2024-04-11:
Updates in PR #1342

kersten1, 2024-04-11:
These look fixed as well.

closed 2024-04-11 by kersten1