riscv/riscv-isa-manual #1345: Zawrs: version 1.01... how about reinterpreting as 1.0.1 or 1.1(.0)?

a4lg, 2024-04-12:
Yes, I know that the latest version of the `Zawrs` extension is version 1.01.

However, considering the nature of the change (fix to a typo), changing the extension version (in the ISA string) to `1p0` to `1p1` seems "too much".  Also, it is the only outlier having the pattern with a zero *prefix* under a point in the current ISA Manual.

I prefer reinterpreting the version 1.01 as "1.0.1" but "1.1(.0)" is definitely an option.  In either case, I don't think keeping "1.01" as-is is a good idea.

Note that, the reason I submitted this issue is, depending on the interpretation of the version, we may have to change the version in the ISA string (as handled in the toolchain).

wmat, 2024-04-12:
Looking at the old repo, Zawrs move to 1.01 with the fix of a typo.  Therefore, 1.1 probably isn't warranted.  However, I'll leave it to @ved-rivos to decide if he'd prefer 1.0.1, which seems in line with versioning best practices.  For reference, here's the old repo: https://github.com/riscvarchive/riscv-zawrs/releases

ved-rivos, 2024-04-12:
1.0.1 sounds fine. However, I am not sure there is a need to include document versions in ISA string parsing - ratified extensions do not change functionally so 1.0.0 vs. 1.0.1 should make no difference.

a4lg, 2024-04-13:
@ved-rivos I agree that but I just needed to confirm.

closed 2025-12-09 by aswaterman