riscv/riscv-isa-manual #1348: Correct user level ISA extension naming convention

atishp04, 2024-04-16:
As discussed on this thread:
https://lists.riscv.org/g/tech-privileged/message/1911

The combined priv spec says
"CSR address space is reserved for a possible future "Sucsrind" extension that extends
indirect CSR access to user mode."

As per Greg, Sucsrind is correct naming as it is a user level ISA extension instead of Unpriv extension.

"The rough distinction being that this is User level specific functionality, whereas Unpriv extensions (e.g. Zi* instructions) can be used at all privilege levels."

This issue is just to confirm that understanding and filed as requested on the mailing list.

aswaterman, 2024-04-16:
What Greg wrote sounds right to me.  So I think the spec says nothing misleading and we can close this, right?

atishp04, 2024-04-20:
Yes. Closing the issue.

closed 2024-04-20 by atishp04