riscv/riscv-isa-manual #1349: Inconsistency for Shift instructions in RV64I

ksungkeun84, 2024-04-16:
Hello,
I've found an inconsisten description about the encoding of shift instructions in RV64I.
In page 36 (V20191213), specifically, it mentions that `"The right shift type is encoded in bit 30."` 
However, in the figure for SLLI/SRLI/SRAI/SLLIW/SRLIW/SRAIW, it seems that bit 30 is used to encode `"Local" or "Arithmetic"` shift.

If it is an actual issue and it needs to be fixed, I would like to contribute to fix it.

aswaterman, 2024-04-16:
Logical vs. Arithmetic is what we mean by shift type. These are indeed the same bit.

ksungkeun84, 2024-04-17:
Hi @aswaterman,
I see. "right" does not means a direction (left or right).

allenjbaum, 2024-04-17:
Actually, "right" does mean a direction. In this case bit30 determines the
*type* of right shift,
which can be either logical (which shifts in zeroes) or arithmetic (which
shifts in sign bits).
There are no different types of left shifts (there are always logical), so
bit30 is only used for right shifts.

ksungkeun84, 2024-04-17:
Thanks @allenjbaum for the clarification :D