riscv/riscv-isa-manual #1350: `zext.w` should not exist in `Zba` group

LeeKaiXuan, 2024-04-17:
Hi, I saw `zext.w rd, rs` was listed in table of Zba extension. However, in ratified version (RISC-V Bit-Manipulation ISA-extensions November 2021), Zba extension should only have 8 instructions without `zext.w`. It seem is a typo and should be removed from the table.
### Current version:
[image]
### Ratified version:
[image]

aswaterman, 2024-04-17:
Duplicate of #1321, resolved by https://github.com/riscv/riscv-isa-manual/commit/7cbf1dcf6ea84958b29a1ce60e9c95a349feb7e7

closed 2024-04-17 by aswaterman

KrystalDelusion, 2024-10-03:
https://github.com/riscv/riscv-isa-manual/commit/7cbf1dcf6ea84958b29a1ce60e9c95a349feb7e7 appears to have been reverted at some point, was this intentional?