riscv/riscv-isa-manual #1355: Rules for CSR and register usage

kersten1, 2024-04-18:
CSR and Register are used somewhat interchangeably. What if we adopted these rules for usage?

- CSR must always be capitalized. 
- First use of CSR in a chapter should include full text plus acronym. Control and State Register (CSR). 
- When plural, lowercase the `s`. CSRs.
- Register is capitalized in title, but lowercase in text. "Supervisor Status (`sstatus`) Register”. But “The `sstatus` register…” 
-Use `CSRs` when referring to a group of registers. Use `Register` when referring to a single register. So "Supervisor CSRs", but "Supervisor Status (`sstatus`) Register”.
- In a title, the format is “Long name (`short name`) Register. All other references in that section can use the short name, but it must be in tics. So "Supervisor Status (`sstatus`) Register”. The rest of the references to that register can be "the `sstatus` register".
- As a general rule, whenever you use a term in tics, it should be followed by what the thing is to help with translation. So avoid statements like this:  “`misa` also helps distinguish different variants of a design.”  And instead use this: “The `misa` register also helps distinguish different variants of a design.”

Thoughts on this? If we get to a consensus, I'll start implementing and enshrine the rules in the dev guide. The majority of the instances in the doc appear to be following these rules.

nick-knight, 2024-04-18:
My personal opinion is (and always has been) that the following text:
> RISC-V defines a separate address space of 4096 Control and Status registers associated with each
hart. This chapter defines the full set of CSR instructions that operate on these CSRs.

hidden in (now) at the beginning of Chapter 7, should be moved to Chapter 1. I think it's fair to trust the reader to read Chapter 1.

allenjbaum, 2024-04-18:
> @nick-knight <https://github.com/nick-knight> Thanks! I updated the word.
>
> About redefining in each section, I was thinking that it might come in
> handy when we start publishing HTML versions as I'm assuming each chapter
> will be on its own HTML page. However, could also link to definition in a
> glossary or else even add hover help that spells it out. I'm open for
> discussion!

You also need to define how field names are formatted, e.g. the  sstatus.SPP
 field

kersten1, 2024-04-18:
@allenjbaum figured I'd take it in chunks. Okay so define CSR in chapter 1 and then first occurrence of an acronym in doc.

kersten1, 2024-04-18:
@kasanovic fair point! so first, how do I know which is a register and what is a specific CSR? Maybe I just need to pay attention to titles. I'm assuming that at least most of the instances of "register" that occur under a heading such as Machine-level CSRs are all CSRs. So looking at this para:

The misa CSR is a WARL read-write register reporting the ISA supported by the hart. This register
must be readable in any implementation, but a value of zero can be returned to indicate the misa
register has not been implemented, requiring that CPU capabilities be determined through a separate non-standard mechanism.

The first instance is correct - The misa CSR - but the instance of "misa register" should actually be "misa CSR"?  What about "This register"? 

Can the titles still use Register as long as they are under the CSR heading? So "Machine ISA (misa) Register" is fine? Just trying to figure out a best course of action here. I recommended using (for example) "misa register" because that seemed to be the most common usage. However, if CSR is more technically correct, then we should definitely move to that. Just might be a more significant change...

Can you confirm?

allenjbaum, 2024-04-19:
The distinction is that "register" usually applies to the 32 general
purpose registers
  (also known as XREGs because they are referenced as X0..X31, and by  ABI*
convention names)
unless prefixed by an "F" instead of an X (for FPR or floating point
registers) or a "V" (for a vector register).
CSRs are technically registers also, but are far more special purpose and
different from each other compared to Xregs (pretend not to notice X0) and
Fregs (pretend not to notice F0 is also the mask) and Vregs (and soon:
Mregs)
"CSR register" is completely redundant, and when referring to them use just
the term "CSR" instead of "register" - so "This register" should be "This
CSR".

*ABI==Application Binary Interface

gfavor, 2024-04-19:
> The distinction is that "register" usually applies to the 32 general
> purpose registers
> (also known as XREGs because they are referenced as X0..X31, and by ABI*
> convention names)
> unless prefixed by [...]

Overall well put.  We have integer, scalar FP, and vector
general "registers", and then we have special "CSRs".  And in a system
there are also memory-mapped registers.  Two of the very examples in the
ISA (versus the many such registers in Non-ISA arch extensions) are the
mtime and mtimecmp registers.

Greg

aamartin0000, 2024-04-19:
minor nit regarding @allenjbaum "... Fregs (pretend not to notice F0 is also the mask) and Vregs":

F0 has no special meaning in the floating-point registers.  V0 is (usually, but not always) a mask register in the vector registers.

allenjbaum, 2024-04-20:
Oops, I blame dyslexia. Yes, I said F0 when I meant V0.

kersten1, 2024-06-05:
moved issue to dev guide

closed 2024-06-05 by kersten1