riscv/riscv-isa-manual #1363: Is it necessary to store ASID in the PTE cache?

Wangrodman, 2024-04-25:
My concern is that different ASIDs may allocate their own page tables at the same physical address. Therefore, I would like to confirm whether the PTE cache needs to match the ASID to differentiate between them.

gfavor, 2024-04-25:
You may need to clarify:  ASIDs allocating their "own" page tables implies _separate_ tables.  But "at the same PA" implies that the same table is being _shared_ by two ASIDs.

In any case, in general one does not have to include ASID in cache tags, but then one has to treat an ASID-specific sfence as a non-ASID-specific sfence.

Wangrodman, 2024-04-30:
Got it, no further question.

Wangrodman, 2024-04-30:
> You may need to clarify: ASIDs allocating their "own" page tables implies _separate_ tables. But "at the same PA" implies that the same table is being _shared_ by two ASIDs.
>
> In any case, in general [...]

closed 2024-04-30 by Wangrodman