riscv/riscv-isa-manual #1366: ISA Specification Consolidation - review comment on volumes 1 and 2 menvcfg register information consistency

dbrash, 2024-04-26:
Regarding the ISA spec consolidation review discussed on this thread:
https://lists.riscv.org/g/tech-privileged/message/1911

volume 2 section 3.1.18 on menvcfg describes the STCE, CBZE, CBCFE and PMM bits as "The definition of the XXXX field will be furnished by the forthcoming Yyyy extension. Its allocation within menvcfg may change prior to the ratification of that extension."

This is one of several instances in the document where these statements need to be updated to reconcile with the ratification of the bits as referenced below:

* volume 1 section 19.5 on the {m,s,h}envcfg CSR definition now defines the CBIE, CBCFE and CBZE bits
*  volume 2 chapter 16 now defines Sstc and the STCE bit field defined in section 16.3.

NOTE: the 2-bit PMM field is awaiting the ratification of the pointer masking extensions which should be a nearterm update to this consolidation effort as the specification (pointer-masking-v1.0.0-rc1.pdf) is frozen/out for final comments.

Finally as a general point, where register bits are described within a register definition such as menvcfg in section 3.1.18, may I suggest we introduce a field/bit-order policy. Using menvcfg as an example FIOM, PBMTE, ADUE, CDE, STCE, CBZE, CBCFE, CBIE, PMM is a bit ad hoc and would be more intuitive if re-ordered to STCE, PBMTE, ADUE, CDE, PMM, CBZE, CBCFE, CBIE, FIOM.

gfavor, 2024-04-26:
On Fri, Apr 26, 2024 at 3:52 AM dbrash ***@***.***> wrote:

> NOTE: the 2-bit PMM field is awaiting the ratification of the pointer
> masking extensions which should be a nearterm update to this consolidation
> effort as the specification (pointer-masking-v1.0.0-rc1.pdf) is frozen/out
> for final comments.
>
The official intent going forward is that a new extension must have its
spec PRs all prepared before either Freeze or Rat-Ready (I forget which),
and then the PRs are merged shortly after the extension is officially
ratified.

> Finally as a general point, where register bits are described within a
> register definition such as menvcfg in section 3.1.18, may I suggest we
> introduce a field/bit-order policy. Using menvcfg as an [...]
What is the reasoning behind your suggested bit reorder?

Also note that ratified arch functionality (including these bits) is not
allowed to be changed.  And note that bits are not simply being allocated
from bottom-up (i.e. there was some conscious thought behind the
assignments as the need for them individually arose over time).

Greg

Message ID: ***@***.***>
>

dbrash, 2024-04-26:
Hi Greg,

My comment was not about bit allocation or any rewording of an
associated bit field's text. I think it is easier if there is consistency
on the order assigned bits are described for all CSRs; it can be high to
low or low to high  bit field order. With 64 bits in a register, the
overall description could span several pages, and I think would make the
bit field descriptions more intuitive to scroll/find as the document grows.

I'm not sure I understand the ratification issue - the bits I mentioned wrt
ratified specs should be fixed now, the other (PMM) I appreciate will need
to wait but just wanted to note it is in effect in a pending state as far
as overall editorial effort is concerned. I am assuming that pointer
masking will be put forward for ratification in the near (wks//mths) term
hence I thought it worth recording.

I applaud the effort going into the consolidation and reducing the number
of documents one needs to track across ratified extensions.

David.

gfavor, 2024-04-26:
I think I now understand your point, which is that the bit descriptions in
the spec should be in natural order, i.e. from low to high or high to low
(and the same order should be used for all CSRs like this).  Yes?

If so, then this would be a good thing for Bill and Kersten to clean up.

Greg

aswaterman, 2024-04-26:
It shouldn't be a hard rule, as there are cases where the narrative flows better if one field is described before another.  In situations where it doesn't matter, then I agree we should not just use some random order.

dbrash, 2024-04-28:
Correct Greg - I'd recommend natural order listing of bit fields as an
editorial cleanup exercise.

David.

dbrash, 2024-04-28:
Apologies Andrew, I didn't notice you had replied as well as Greg. Where
there is a case for a hierarchy/grouping that is different from all in
natural order that could be highlighted/used.

David.

aswaterman, 2024-04-29:
I don’t agree that we even need to highlight the deviation from natural order—-it’s not like this is an ISA-visible property. We should just avoid deviating from natural order unless there’s a good reason.

aswaterman, 2025-12-09:
Closing as answered.

closed 2025-12-09 by aswaterman