riscv/riscv-isa-manual #1368: Reading of Unprivileged Architecture Version 20240411, Introduction

HolgerBlasumSYSGO, 2024-04-27:
p. 2 "The document contains the following versions of the RISC-V ISA modules" --> would be a perfect place to give a short rationale why the ISA modules / extensions are named and ordered this way.

p. 12 "The retirement of an instruction" -> this is RISC-V jargon, better "The execution of an instruction"

p. 12 "hardware thread (hart) and a software thread context", why is one of them called "context" and the other not.

p. 15 "Executing each RISC-V" -> first line indented inconsistently on left side and also additional blank on right side

p 16 All the 32-bit instructions in the base ISA have their lowest two bits set to "11". The optional compressed 16-bit instruction-set extensions have their lowest two bits equal to 00, 01, or 10. --> Inconsistent formatting of "11" and "00" etc

p 23, "R-Type", "I-Type", "S-Type", "U-Type" -> explain meaning of mnemonics in a comment 

p 31 "Sign extensions always uses inst[31]. Sign-extension is one of the most critical operations on immediates" --> correct to "always use int". Either use "Sign extension" or "Sign-extension"

aswaterman, 2024-04-27:
@HolgerBlasumSYSGO Thanks for the feedback.  At the moment, we are focused on fixing transliteration errors that cropped up during the AsciiDoc transition.  It helps us out if you file separate issues for transliteration errors vs. new editorial suggestions.

> would be a perfect place to give a short rationale why the ISA modules / extensions are named and ordered this way

The Intro provides this rationale.  I don't think we should migrate that into the Preface, the purpose of which is to serve as a changelog.

> this is RISC-V jargon

No, this is standard computer-architecture terminology.  In this particular context, replacing "retire" with "execute" makes the meaning less clear, since we are talking about the final outcome.

> first line indented inconsistently

Fixed.  Thanks for pointing this out.

> Inconsistent formatting

Fixed, thanks.

> "R-Type", "I-Type", "S-Type", "U-Type" -> explain meaning of mnemonics in a comment

Do other folks agree this is insufficiently clear from context?

> correct to "always use int"

"inst" is correct as-is.

> Either use "Sign extension" or "Sign-extension"

Indeed we should not hyphenate compound nouns (but of course we should hyphenate compound adjectives).

aswaterman, 2024-04-27:
Resolved by 6d137a77e7e8619bf6eaf36a27674ec5bea08b21, 3d703f3f2d8fb5ebdbc8ee6be508661c318d451b, and c8773d5ad3b6d2b4bb92ed0c5ea26abac5b52c68.

closed 2024-04-27 by aswaterman

HolgerBlasumSYSGO, 2024-04-29:
Hi Andrew,

thanks for following up.

 >     would be a perfect place to give a short rationale why the ISA 
modules / extensions are named and ordered this way
 >
 > The Intro provides this rationale. I don't think we should migrate 
that into the Preface, the purpose of which is to serve as a changelog.

I agree that the preface is the wrong place. Then perhaps put this into 
Section 1.8 which would explain e.g. that the "Zi" in "Zifencei" is some 
prefix (because letter "Z" was still free I assume) and that the 
non-alphabetical ordering of extensions reflects some other grouping idea.

 >> Page 12, "The following elements constitute forward progress: 
retirement of an instruction" -> this is RISC-V jargon
 > No, this is standard computer-architecture terminology. In this 
particular context, replacing "retire" with "execute" makes the meaning 
less clear, since we are talking about the final outcome.

Maybe some text like "retirement of an instruction, after all its 
microoperations have been executed without raising a trap".

I concede "retirement" is standard computer-architecture jargon, but 
here we are on the third page of the highest-level introduction: 
everything before was completely readable without being a 
computer-architecture specialist, and that was first place where it 
became less self-contained (the concept or term of "retirement" is not 
explained in the document). Mentioning execution also would be nicer 
linkage to "execution environment" in the sentence preceding it.

[...]

Best