riscv/riscv-isa-manual #1376: Zcmp depends on ABI register names, which are no longer introduced in unprivileged manual

christian-herber-nxp, 2024-04-30:
The ABI names of reigsters are no longer defined in the specification.
There are however parts of the ISA that depend on the ISA.
E.g. C introduces SP relative loads/stores, but there it is specified that X2 is used explicitly.

Zcmp on the other hand does not make that explicit, but refers to the ABI names only. 
Without the ABI definition, an implementer would not know which registers to push/pop.

A resolution would be to add the ABI definition back into the manual.

aswaterman, 2024-04-30:
Referencing the C extension chapter would suffice, since this information is spelled out there (and since Zcmp builds upon C).

ThinkOpenly, 2026-02-26:
Will a fix for this also remove the handful of references to "UABI" and "EABI" that have no definition?
```
$ grep -E 'UABI|EABI' $(find src -type f)
src/zc.adoc:_rlist_ values 0 to 3 are reserved for a future EABI variant called _cm.push.e_
src/zc.adoc:All ABI register mappings are for the UABI. An EABI version is planned once the EABI is frozen.
src/zc.adoc:_rlist_ values 0 to 3 are reserved for a future EABI variant called _cm.pop.e_
src/zc.adoc:All ABI register mappings are for the UABI. An EABI version is planned once the EABI is frozen.
src/zc.adoc:_rlist_ values 0 to 3 are reserved for a future EABI variant called _cm.popretz.e_
src/zc.adoc:All ABI register mappings are for the UABI. An EABI version is planned once the EABI is frozen.
src/zc.adoc:_rlist_ values 0 to 3 are reserved for a future EABI variant called _cm.popret.e_
src/zc.adoc:All ABI register mappings are for the UABI. An EABI version is planned once the EABI is frozen.
src/zc.adoc:The _s_ register mapping is taken from the UABI, and may not match the currently unratified EABI. _cm.mvsa01.e_ may be included in the future.
src/zc.adoc:The _s_ register mapping is taken from the UABI, and may not match the currently unratified EABI. _cm.mva01s.e_ may be included in the future.
```