riscv/riscv-isa-manual #1377: Clarify PBMT and PBMTE rule when implicit memory access for VS-stage address translation

Wangrodman, 2024-04-30:
Although the leaf PTE is not a stage leaf PTE, the specification does not clarify whether PBMT is meaningful in this scenario. 
Please clarify the following two questions:
1. Does the leaf PTE of implicit memory access for VS-stage address translation affect the memory attribute of accessing the guest's PTE?
2. If the answer to Q1 is YES, which PBMTE, menvcfg or henvcfg, controls the PBMT?"

gfavor, 2024-05-01:
#1.  The last paragraph of the Svpbmt spec specifies the behavior for two-stage translations.  In general PBMTs don't affect the PMAs for page tables and the implicit accesses to them.  They only affect the final explicit access after the access's VA has been translated to a PA.

Wangrodman, 2024-05-13:
> #1. The last paragraph of the Svpbmt spec specifies the behavior for two-stage translations. In general PBMTs don't affect the PMAs for page tables and the implicit accesses to them. They only affect the final explicit access after the access's VA has been translated to a PA.

Got it, no further question.

gfavor, 2024-10-15:
For henvcfg.PBMTE the spec says "When PBMTE=1, Svpbmt is available for VS-stage address translation. When PBMTE=0, the
implementation behaves as though Svpbmt were not implemented for VS-stage address translation."

And for menvcfg .PBMTE the spec says "The PBMTE bit controls whether the Svpbmt extension is available for use in S-mode and G-stage address translation (i.e., for page tables pointed to by satp or hgatp). When PBMTE=1, Svpbmt is available for Smode and G-stage address translation."

In other words, henvcfg.PBMTE controls only VS-stage PBMT, while menvcfg.PBMTE controls both G-stage and VS-stage (i.e. when vsatp substitutes for satp while V=1).  This is also reflected in the following excerpt from later in the spec: "Similarly, if the setting of the PBMTE bit in menvcfg is changed, an HFENCE.GVMA instruction with rs1=x0 and rs2=x0 suffices to synchronize with respect to the altered interpretation of G-stage and VS-stage PTEs' PBMT fields."

Lastly, Svpbmt does not apply to implicit accesses (nothing in the spec explicitly says otherwise).  This is reflected by the last paragraph in the Svpbmt spec that only talks about determining the final attributes that apply to the explicit access being translated:

"When two-stage address translation is enabled within the H extension, the page-based memory types are
also applied in two stages. First, if hgatp.MODE is not equal to zero, non-zero G-stage PTE PBMT bits
override the attributes in the PMA to produce an intermediate set of attributes. Otherwise, the PMAs serve
as the intermediate attributes. Second, if vsatp.MODE is not equal to zero, non-zero VS-stage PTE PBMT
bits override the intermediate attributes to produce the final set of attributes used by accesses to the page
in question. Otherwise, the intermediate attributes are used as the final set of attributes."

Wangrodman, 2024-10-16:
Hi @gfavor ,
"Svpbmt does not apply to implicit accesses."

There are two PBMT illegal conditions:
PBMT = 3 (reserved encoding)
PBMT = 1 or 2 but PBMTE = 0
How should the above conditions be handled? Should they be raised or suppressed?

gfavor, 2024-10-16:
The specs for the *envcfg PBMTE bits specify that if Svpbmt is disabled for a stage of translation, then the implementation behaves as though Svpbmt were not implemented (for that stage of translation).

For reserved PBMT encodings the Svpbmt spec specifies taking a page fault.

terranfund, 2024-10-16:
Hi @gfavor 
Thanks for your kindly reply.
> In other words, henvcfg.PBMTE controls only VS-stage PBMT, while menvcfg.PBMTE controls both G-stage and VS-stage
Do you want to express the following SPEC statement?
I am assuming you had some typo.
> The menvcfg.PBMTE bit controls whether the Svpbmt extension is available for use in **S-mode and G-stage** address translation (i.e., for page tables pointed to by satp or hgatp).

BTW,
As your state in previous post:
> Lastly, Svpbmt does not apply to implicit accesses
Does it mean it is Svpbmt is not implemented for implicit access, subject to G-stage address translation, for VS-stage address translation?
If yes, following the SPEC says,
> If Svpbmt is not
implemented, bits 62-61 remain reserved and must be zeroed by software for forward compatibility, or
else a page-fault exception is raised
I wonder **If we can say PBMT=1/2/3 on leaf PTEs along with implicit access, subject to G-stage address translation, for VS-stage address translation raise a guest page-fault?**
And following the SPEC statement:
> For non-leaf PTEs, bits 62-61 are reserved for future standard use. Until their use is defined by a
standard extension, they must be cleared by software for forward compatibility, or else a page-fault
exception is raised.
I wonder **If we can say PBMT= 1/2/3 on non-leaf PTEs along with implicit access, subject to G-stage address translation, for VS-stage address translation raise a guest page-fault?**

gfavor, 2024-10-16:
Yes, I misspoke.  henvcfg.PBMTE controls VS-stage PBMT availability, while menvcfg.PBMTE controls S-mode and VS-stage PBMT availability.

The question of applicability to implicit accesses is orthogonal to the type of fault taken when non-zero PBMT is encountered and the corresponding PBMTE bit is 0.  The presence or absence of non-zero PBMT and of the Svpbmt extension does not affect implicit accesses and the memory type of implicit accesses is only determined by the PMAs.

Conceptually, from the perspective of a guest OS, it is managing VS-stage PBMT to control the memory type of explicit memory accesses.  The PBMT in the VS-stage PTE controls the memory type of the translated explicit access, and does not control the memory type of any tablewalk implicit accesses.  And of course it is unaware of there being any two-stage translation.  So it would be strange for that leaf PBMT to have affect over G-sage implicit accesses.

But as the spec says for an overall two-stage tablewalk to translate the address of an explicit access, if during a given stage of translation either a reserved PBMT encoding is encountered, or a non-zero encoding is encountered and Svpbmt is effectively not implemented for that stage, then a page-fault exception is raised.

But, when raising a page-fault in connection with G-stage translation, what is actually raised is a guest-page-fault.  As the H spec says:  "guest-page-fault exceptions are raised instead of regular page-fault exceptions."

terranfund, 2024-10-16:
@gfavor 
Please confirm the following two questions whether they are both true?
1. If we can say PBMT=1/2/3 on leaf PTEs along with implicit access, subject to G-stage address translation, for VS-stage address translation should raise a guest page-fault **without referring to any PBMTE(of menvcfg or henvcfg)**?
2. If we can say PBMT= 1/2/3 on non-leaf PTEs along with implicit access, subject to G-stage address translation, for VS-stage address translation should raise a guest page-fault **without referring to any PBMTE(of menvcfg or henvcfg)**?

gfavor, 2024-10-17:
The terminology "implicit G-stage" and "PTEs along with implicit access" are not present in the arch specs and are subject to (mis)interpretation.  In any case (and this is back-pedaling from part of what I said earlier), I think the following answers the original and follow-on questions in this issue (excluding the *envcfg-related questions and answers which still stand).

- The VS-stage leaf PTE's PBMT does **not** affect any implicit PTE memory accesses (versus other types of implicit memory accesses).
- The G-stage leaf PTE's PBMT that is used to translate a VS-stage implicit PTE access **does** affect the implicit memory access of that PTE (in combination with the PMAs for that access).
- The G-stage leaf PTE's PBMT that is used as part of a two-stage translation of an explicit access's address **does** affect the explicit memory access as specified by the Svpbmt spec for determining the final attributes of the access.
- Non-zero PBMT bits and use of the Reserved PBMT encoding, as defined in the arch spec and based on *envcfg.PBMTE bits, results in a page-fault when encountered in VS-stage PTEs, and a guest-page-fault when encountered in G-stage PTEs.

Hopefully this resolves the outstanding questions.

gfavor, 2024-10-17:
Fyi, I am working with Andrew Waterman to create a clarification to the Svpbmt spec that more explicitly covers Svpbmt wrt VS-stage implicit PTE memory accesses.

terranfund, 2024-10-17:
Hi @gfavor 
The term I used in this thread refers to the text in section 10.15. Two-stage Address Translation, RISC-V Privileged Version 20240411:

> When V=1, memory accesses that would normally bypass address translation are subject to G-stage
address translation alone. This includes memory accesses made in support of VS-stage address
translation, such as reads and writes of VS-level page tables.

I question were based on such memory accesses:

> 1. If we can say PBMT=1/2/3 on leaf PTEs along with implicit access, subject to G-stage address translation, for VS-stage address translation should raise a guest page-fault without referring to any PBMTE(of menvcfg or henvcfg)?

> 2. If we can say PBMT= 1/2/3 on non-leaf PTEs along with implicit access, subject to G-stage address translation, for VS-stage address translation should raise a guest page-fault without referring to any PBMTE(of menvcfg or henvcfg)?

Or allow me to revise the questions from confusing since they didn't have formal terms to describe:

1. If we can say PBMT=1/2/3 of leaf PTEs on memory access, subject to G-stage address translation, made in support of VS-stage address translation should raise a guest page-fault without referring to any PBMTE(of menvcfg or henvcfg)?
2. If we can say PBMT= 1/2/3 of non-leaf PTEs on memory access, subject to G-stage address translation, made in support of VS-stage address translation should raise a guest page-fault without referring to any PBMTE(of menvcfg or henvcfg)?

gfavor, 2024-10-17:
I'm not 100% certain I fully understand the exact questions being asked, but let's see if this answers them:

If PBMTE=0 for a given stage of translation, then the behavior is as if Svpbmt is not implemented.  Which means a page-fault for a non-zero VS-stage PBMT field, and a guest-page-fault for a non-zero G-stage PBMT field

And if PBMTE=1, then it's the same story as far as fault type, but for a more constrained set of PBMT field cases.

terranfund, 2024-10-17:
@gfavor 
Please be informed, here, that originating questions are concerned about PBMT=1/2 or 3 on leaf PTEs.
Hopefully it helps.

gfavor, 2024-10-17:
Yes, and the arch spec directly specifies what happens fault-wise for PBMT on leaf PTEs when the relevant PBMTE= 0 or 1.  And for PBMT=1/2 it directly specifies what happens during a two-stage translation.  And my post https://github.com/riscv/riscv-isa-manual/issues/1377#issuecomment-2418234399 specified what happens for G-stage leaf PTE PBMT that is used to translate a VS-stage implicit PTE access.

What am I missing that is unanswered?

terranfund, 2024-10-17:
@gfavor 
I am assuming the following text in https://github.com/riscv/riscv-isa-manual/issues/1377#issuecomment-2418234399 contradict the texts you had answered

> The G-stage leaf PTE's PBMT that is used to translate a VS-stage implicit PTE access does affect the implicit memory access of that PTE (in combination with the PMAs for that access).

contradicts 

> Lastly, Svpbmt does not apply to implicit accesses (nothing in the spec explicitly says otherwise). This is reflected by the last paragraph in the Svpbmt spec that only talks about determining the final attributes that apply to the explicit access being translated:

and 

> The last paragraph of the Svpbmt spec specifies the behavior for two-stage translations. In general PBMTs don't affect the PMAs for page tables and the implicit accesses to them. They only affect the final explicit access after the access's VA has been translated to a PA.

gfavor, 2024-10-17:
In https://github.com/riscv/riscv-isa-manual/issues/1377#issuecomment-2418234399, when I said "In any case (and this is back-pedaling from part of what I said earlier), ..." I was essentially saying to ignore any earlier statement that appeared to be contradictory to what I proceeded to say in https://github.com/riscv/riscv-isa-manual/issues/1377#issuecomment-2418234399.

Wangrodman, 2024-10-17:
> Fyi, I am working with Andrew Waterman to create a clarification to the Svpbmt spec that more explicitly covers Svpbmt wrt VS-stage implicit PTE memory accesses.

Thanks, it's helpful for developers. Please paste the commit link for the riscv-isa-manual after you finish.

gfavor, 2024-10-18:
Closing this issue since it sounds like the questions have been resolved and a clarification has been added to the Svpbmt spec (see Andrew's PR).

closed 2024-10-18 by gfavor

terranfund, 2024-10-18:
@gfavor @Wangrodman 
Thansk for your great support. For me, it clarified all questions throughout the thread, including the quesiton @Wangrodman makde at the first of it.

aswaterman, 2024-10-19:
We're glad to have helped!

terranfund, 2024-11-27:
Hi @aswaterman 
Due the latest PR supplement

> These final attributes apply to implicit and explicit accesses that are subject to both stages of address translation. For accesses that are not subject to the first stage of address translation, e.g. VS-stage page-table accesses, the intermediate attributes apply instead.

I wonder whether the following statement needs adjustment as well.

> Similarly, if the setting of the PBMTE bit in menvcfg is changed, an HFENCE.GVMA instruction with rs1=x0 and rs2=x0 suffices to synchronize with respect to the altered interpretation of G-stage and VS-stage PTEs PBMT fields.

Because the intermediate attributes (from G-stage translation) apply to VS-stage page-table accesses, when menvcfg.PBMTE is changed, an HFENCE.GVMA doesn't suffice to synchronize with respect to the altered interpretation of G-stage PTE's PBMT filed.
**For this case, an extra HFENCE.VVMA, if needed, with rs1=x0 and rs2=x0 is needed to synchronize overall VS-stage translation, composed of VS-stage page-table accesses affected by the intermediate attributes.**

> Conceptually, an implementation might contain two address-translation caches: one that maps guest virtual addresses to guest physical addresses, and another that maps guest physical addresses to supervisor physical addresses. HFENCE.GVMA need not flush the former cache, but it must flush entries from the latter cache that match the HFENCE.GVMA’s address and VMID arguments.

terranfund, 2024-11-28:
Hi @aswaterman 
The fact that attributes used for VS-stage page-table accesses have changed will change the effect of VS-stage page-table accesses. For example, attributes used for VS-stage page-table accesses changed are of reserved value (0/1/2) when menvcfg.PBMTE is changed from 1 to 0.

aswaterman, 2024-12-04:
@terranfund I agree that not executing an HFENCE.GVMA instruction in the scenario you mentioned might be semantically visible, but it corresponds to a scenario that is uninteresting in practice.  Properly behaved M-mode software will never change the value of PBMTE from 1 to 0 under the feet of an OS or hypervisor.

I think the semantics of these instructions are quite clear from the normative text, and providing extra non-normative guidance to software that discusses situations that would never occur in proper operation is not helpful.

terranfund, 2024-12-04:
Hi @aswaterman 

> The fact that attributes used for VS-stage page-table accesses have changed will change the effect of VS-stage page-table accesses. For example, attributes used for VS-stage page-table accesses are of reserved value (1/2/3) when menvcfg.PBMTE is changed from 1 to 0 (from having no guest page-fault to having guest page-fault). HFENCE.GVMA need not flush the cache entries that cached guest virtual addresses to guest physical addresses.

The following question is the originating one I wondered, since **HFENCE.GVMA** need not flush the cache entries that cached guest virtual addresses to guest physical addresses:
For this case, an extra **HFENCE.VVMA**, as needed, with rs1=x0 and rs2=x0 is needed to synchronize overall VS-stage translation, composed of VS-stage page-table accesses affected by the intermediate attributes.

gfavor, 2024-12-04:
I'm a little confused by the missing of questions about PBMT affecting VS-stage page table accesses, PBMT affecting VS-stage translations, and G-stage PBMT affecting VS-stage PTE accesses.  So ... 

To the statement above saying "For this case, an extra HFENCE.VVMA, as needed, with rs1=x0 and rs2=x0 is needed to synchronize overall VS-stage translation, composed of VS-stage page-table accesses affected by the intermediate attributes.", changes to G-stage translations affect VS-stage PTE accesses but do NOT affect VS-stage translations.  So there would be no need for an HFENCE.VVMA.

terranfund, 2024-12-05:
Because the **intermediate attributes** (from the G-stage translation the memory access made in support of VS-stage address translation is subject to) apply to VS-stage page-table accesses, when menvcfg.PBMTE is changed, an _HFENCE.GVMA_, obeying current SPEC, doesn't suffice to synchronize with respect to the altered interpretation of G-stage PTE's PBMT field(above-mentioned **intermediate attributes**), since HFENCE.GVMA need not flush the cache entries that cached guest virtual addresses to guest physical addresses(the VS-stage address translation).
Thus, to reflect the influences of menvcfg.PBMTE on the **intermediate attributes**, I think an extra _HFENCE.VVMA_, in addition to _HFENCE.GVMA_, is needed actually.

gfavor, 2024-12-06:
As already stated, the intermediate attributes affect how PTE accesses are performed, but they do not affect VS-stage translations themselves (and the values returned by PTE accesses from the addressed memory locations).

terranfund, 2024-12-06:
OK. Thanks for your great insights while we did care that intermediate attributes affect how PTE accesses are performed in our platform. We will note this.

terranfund, 2025-01-08:
Tangible example:
Old intermediate atrributes equals to 1 (NC).
Old menvcfg.PBMTE equals to 1.
Assumed no guest page fault is detected so far.
When menvcfg.PBMTE is changed to 0, only HFENCE.GVMA flush-all is required but HFENCE.VVMA accordingly, should guest page-fault be detected?

gfavor, 2025-01-08:
What would be the cause of a guest page fault in this scenario?

gfavor, 2025-01-10:
On the one hand it is correct that menvcfg.PBMTE=0 results in any non-zero PBMT field in G-stage PTEs causing a guest-page fault.  But this is all orthogonal to VS-stage translations, caching of such translation information, and the use of HFENCE.VVMA to manage changes to VS-stage page tables.

In other words, changes to G-stage translations, including resulting from changing menvcfg.PBMTE, do not affect VS-stage translations.  At most (via Svpbmt), they only affect how VS-stage PTE reads are performed, but not the PTE values read and/or how they are interpreted.  And they only affect the second stage of two-stage address translations.