riscv/riscv-isa-manual #1379: Does misaligned exception in nottaken branches cause an exception?

omerguzelelectronicguy, 2024-05-02:
I couldn't understand the misaligned exception when the branch is not taken and the target has a misaligned address. Is there any explanation in the spec? If not I think it should be added to the spec.

stnolting, 2024-05-02:
This is from an older release of the spec:

[image]

So, no, there will be no exception in that case.

closed 2024-05-02 by aswaterman

omerguzelelectronicguy, 2024-05-06:
Thank you. I think it should be in the latest version of the spec too.

aswaterman, 2024-05-06:
> it should be in the latest version of the spec

It is: https://github.com/riscv/riscv-isa-manual/blob/87edab70ddd2fedc6d01603a65f393163025633f/src/rv32.adoc#L648-L653