riscv/riscv-isa-manual #1384: How to clear hvip.VSEIP when hvip.VSEIP is asserted and VS-level exteranl interrupt is delegated to VS-mode？

jinyinghan324, 2024-05-06:
If VS-level external interrupt is asserted because hvip.VSEIP is asserted and VS-level exteranl interrupt is delegated to VS-mode, then interrupt hanlder runs in VS-mode and hvip cannot be written.  How to clear hvip.VSEIP in handler？

aswaterman, 2024-05-09:
Thinking about how this works without virtualization: normally the supervisor external interrupt handler would claim an interrupt from the PLIC by doing an MMIO access, which as a side effect would cause the PLIC to deassert the interrupt line, making `sip.SEIP` become 0.

The analogous thing with virtualization would be that the hypervisor is emulating a PLIC for the guest OS.  When the guest OS executes the MMIO access to claim an interrupt from the PLIC, that MMIO access will trap into the hypervisor so that it can emulate the access.  The hypervisor's emulation routine then has the opportunity to clear `hvip.VSEIP`.

aswaterman, 2024-05-09:
I should add: I think this is _only_ interesting in the case that the hypervisor is emulating an interrupt controller for the guest.  If the guest has access to a real interrupt controller, then `hvip.VSEIP` would not be used, and so there would be no need to clear it.

Steven-Li-Xiaogang, 2025-01-10:
@aswaterman
1. it seems the privilege spec implicit that external interrupt caused pending would have a sperate register named 'E', vsip.VSEIP/vsip.VSEIP was called register 'B', and SW only was able to change the status of register 'B', so it seems that we could not have a way to clear hvip.VSEIP pending status with MMIO access to claim/clear interrupt from interrupt controller (such as PLIC or IMSIC).
2. SW also can not recognize it's a virtual interrupt injected by writing hvip.VSEIP or a real external interrupt from controller while they arrived at the same time, so we could not issue an ecall to let HS/M-mode help to clear hvip.VSEIP like what you have mentioned in https://github.com/riscv/riscv-isa-manual/issues/1589

closed 2025-01-10 by aswaterman