riscv/riscv-isa-manual #1385: does Xtvec need to support all invalid adddresses?

tariqkurd-repo, 2024-05-07:
`If mtvec is writable, the set of values the register may hold can vary by implementation.`

`The BASE field in stvec is a field that can hold any valid virtual or physical address, subject to the following alignment constraints`

`mepc is a WARL register that must be able to hold all valid virtual addresses. It need not be capable of holding all possible invalid addresses.`

`sepc is a WARL register that must be able to hold all valid virtual addresses. It need not be capable of holding all possible invalid addresses.`

`If mtval is not read-only zero, it is a WARL register that must be able to hold all valid virtual addresses and the value zero. It need not be capable of holding all possible invalid addresses.`

`stval is a WARL register that must be able to hold all valid virtual addresses and the value 0. It need not be capable of holding all possible invalid addresses`

I'm trying to understand whether `mtvec/stvec` can be trimmed to only store valid virtual addresses.

The `mtvec` statement about is a bit amiguous and doesn't match the `stvec` statement, I don't know any they would be different.

The other registers: `mepc, sepc, mtval, stval` are specified consistently.

Can we have the same text for all of them? Is there a reason why `mtvec/stvec` need to be able to store more address bits than the others?

pdonahue-ventana, 2024-05-07:
My understanding is that mtvec can be hardwired to one specific value if you want (e.g. the handler is in ROM).  So it doesn't need to support all invalid addresses or even all valid addresses.  The others need to support all valid addresses.

aswaterman, 2024-05-07:
> I'm trying to understand whether mtvec/stvec can be trimmed to only store valid virtual addresses.

Yes, because of the first sentence you quoted: "If mtvec is writable, the set of values the register may hold can vary by implementation."

> Can we have the same text for all of them?

No, the constraints on `mepc` and `mtval` are inherently a little different from `mtvec`.  Indeed it's not useful for `mtvec` to be able to hold invalid addresses.  But it is important that `mepc` and `mtval` hold at least one invalid address, to indicate the reason for an exception is that the address is invalid (as opposed to being valid but currently faulting).

tariqkurd-repo, 2024-05-08:
ok - thanks - so the rationale is that `mtvec` doesn't need to be able to hold _any_ invalid addresses whereas `mepc` and the others must store _at least_ one.

aswaterman, 2024-05-08:
A fine summary of the discussion.

closed 2025-12-09 by aswaterman