riscv/riscv-isa-manual #1386: Realign bytefield images

kersten1, 2024-05-07:
Compare bytefield images from latex version of docs to asciidoc version and fix any inconsistencies.

You can find the original latex version of the doc here - https://drive.google.com/file/d/1s0lZxUZaa7eV_O0_WsZzaurFLLww7ou5/view

and https://drive.google.com/file/d/1EMip5dZlnypTk7pt4WWUKmtjUKTOkBqh/view

Latest asciidoc versions can be found at top of tree - https://wiki.riscv.org/

M-S-10, 2025-01-15:
Hey @kersten1, I am starting to fix this issue. Can you assign this to me?

kersten1, 2025-01-15:
@M-S-10 let's see what you can do. Are you a member of RISC-V?

M-S-10, 2025-01-15:
I just joined as an individual member, waiting for me confirmation email. :)

M-S-10, 2025-01-24:
Hi, I have reviewed the bytefield images (Volume 1 & Volume 2) and identified discrepancies, including alignment issues and annotation inconsistencies. My next steps this week will focus on resolving these inconsistencies to ensure clarity and alignment across the manual. I will update you with my progress or any challenges encountered.

Please let me know if there are specific priorities or additional guidelines to consider while making these corrections.

Here are my findings: 

[image]

kersten1, 2025-12-10:
@M-S-10 I'm sorry! are you still around and interested in doing this work?